openthread/openthread #6269: Openthread video guide or tutorial?

Hard1kPrajapat1, 2021-03-11:
Hello Team,

I am new to this technology, and I want to adopt the tutorial or any video lecture for the Openthread to learn. I have searched the same on Udemy google but could not found. Is there any link or websites where we can have and learn?

Thanks,
Hardik

jwhui, 2021-03-11:
There are a number of guides and tutorials linked on https://openthread.io/

Hard1kPrajapat1, 2021-03-11:
Thanks for sharing the link, there are a lot of guides available which will be useful. But I am looking for the Video tutorial which explains the openthread from scratch, its connectivity, and working.

jwhui, 2021-03-11:
> Thanks for sharing the link, there are a lot of guides available which will be useful. But I am looking for the Video tutorial which explains the openthread from scratch, its connectivity, and working.

There is a video from Google I/O 2018 on the front page of https://openthread.io/ that provides an overview of Thread and OpenThread.

Hard1kPrajapat1, 2021-03-11:
Okay, thank you so much.

closed 2021-03-11 by Hard1kPrajapat1