openthread/openthread #6278: How to create thread or loop?

purvisb, 2021-03-14:
I'm working out of the nRF SDK on the NRF52840-mdk.

I would like to read from a sensor periodically with a set time increment. How should I do this? 

I did experiment with an bsp. But that yielded strange results. Does OpenThread have some API I can use? Or is an RTOS better suited for this use case?

jwhui, 2021-03-15:
If you want more general system services like schedulers, timers, and peripheral drivers, you should look to SDKs and/or RTOSes. OpenThread's primary scope is to provide Thread networking.

purvisb, 2021-03-18:
@jwhui thank you for the clarification. Out of curiosity, what are [tasklets?](https://openthread.io/reference/group/api-tasklets) To me this sounds kind of like a (OS) thread. Or is this just used exclusively to coordinate openthread?

jwhui, 2021-03-18:
> @jwhui thank you for the clarification. Out of curiosity, what are [tasklets?](https://openthread.io/reference/group/api-tasklets) To me this sounds kind of like a (OS) thread. Or is this just used exclusively to coordinate openthread?

Tasklets are a common OS concept. OpenThread has its own implementation of tasklets for scheduling purposes within OpenThread itself.

closed 2021-03-18 by purvisb