openthread/openthread #6283: WPANTUND fails to initialize on-board nrf52840

ChauhanPD, 2021-03-15:
I am facing the same issue (as reported in #6003) with wpantund as well as NCP device. I have taken the latest commit of wpantund, but still got the same error.
In my set-up, I have an on-board nrf52840 chip connected via UART with the host processor (i.MX6ULL) on our custom board.

I have flashed nrf52840 Chip with cli-ftd.hex file which is generated with below command:

make -f examples/Makefile-nrf52840 clean
make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 UDP_PROXY=1 BORDER_AGENT=1 LINK_RAW=1 SERVICE=1

and flashed via j-link using below command :
nrfjprog -f nrf52 --chiperase --program output/nrf52840/bin/ot-cli-ftd.hex --reset

I have taken out the RX, TX, CTS, RTS, and GND from the nrf52840 UART which is connected to the host processor and connected with the PC (ubuntu) to give the command from the PC terminal.

I have gone through many posts/forums, but no progress for the last two days. I think my set-up is different from all users as I am not using any mdk/sdk or dongle of nrf52840. Let me again clarify my set-up. I have a custom board having i.MX6ULL as a host processor and it is connected to nrf52840 via UART. I have taken the same 4 lines (TX, RX, CTS and RTS) and connected to the FTDI cable to access nrf52840 using cli application.

When I open terminal for nrf52840, there is no ">" observed and also no response for "state" and "reset' command.

My first task is to run ot-br on host processor using nrf5840 as radio processor. I can open the web-page of open thread in android mobile phone, but not able to create the OpenThread network. For that I checked and found that wpantund service is required to run and status of "wpanctl" commnad must be "initialized". But the output of executing wpantund is as shown below:

/usr/sbin/wpantund -o Config:NCP:SocketPath "serial:/devv/ttymxc0,raw,b115200,crtscts=1" -o Config:TUN:InterfaceName utun7
wpantund[24506]: Starting wpantund 0.08.00d (Mar 11 2021 09:57:59) . . .
wpantund[24506]: 	SOURCE_VERSION = 0.07.01-354-ge8c110e
wpantund[24506]: Configuration file "/etc/wpantund.conf" read.
wpantund[24506]: Ready. Using DBUS bus ":1.12"
wpantund[24506]: Running as root without dropping privileges!
wpantund[24506]: ../../../git/src/ncp-spinel/SpinelNCPInstance-Protothreads.cpp:412: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[24506]: Resetting and trying again... (retry 1)
wpantund[24506]: ../../../git/src/ncp-spinel/SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[24506]: Resetting and trying again... (retry 2)
wpantund[24506]: ../../../git/src/ncp-spinel/SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[24506]: Resetting and trying again... (retry 3)
wpantund[24506]: ../../../git/src/ncp-spinel/SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[24506]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[24506]: State change: "uninitialized" -> "uninitialized:fault"
^C
Caught SIGINT!
wpantund[24506]: select() errno="Interrupted system call" (4)
wpantund[24506]: Cleaning up. (gRet = 130)
wpantund[24506]: Stopped.

Kindly help / guide me to successfully run the wpantund service and be able to create the OpenThread network.

**To Reproduce** Information to reproduce the behavior, including:
1. Git commit id
    wpantund:   e8c110e7389446a6b133389c5572c183a095388b  ( 10 Mar, 2021)
    openthread: 93b6fea3aa10e26f17ad2b877f002ca2cd81b745 ( 5 Mar, 2021)
    ot-br:            36007c3fa33d605b926d9ac387fec9abb128e594 (13 Aug, 2019)

2. IEEE 802.15.4 hardware platform
     Custome board having i.MX6ULL and on-board nrf52840.

**Expected behavior** A clear and concise description of what you expected to happen.
wpantund service should communicate with on-board nrf52840 and I should be able to create the OpenThread network.

abtink, 2021-03-15:
@ChauhanPD , I can suggest one thing to check based on the error logs you posted.

Would suggest to check if the socket path you are using and passing to `wpantund` in 
```
Config:NCP:SocketPath "serial:/devv/ttymxc0,raw,b115200,crtscts=1"
```
 is correct and all parameter/configs are also correct and match the configs on the NCP firmware build.
- The error from `wpantund` indicates that it cannot talk to the NCP (no responses from NCP) on the provided socket path.

ChauhanPD, 2021-03-16:
Thanks, @abtink for your reply.

Actually, I was giving "/dev/ttymxc0" only as a socket path. But the captured log had "/devv/ttymxc0"!
Anyway, I have again given the correct path. Also, re-compiled the NCP firmware and flashed on-board nrf52840 using j-link / nrfConnect utility.
But the output is the same as my first post.

Should I flash NCP firmware or RCP firmware in on-board nrf52840 with the github code of ot-br-posix?
Also, is wpantund dependent on any service?

abtink, 2021-03-16:
In general, `wpantund` works with an NCP (so NCP build and not RCP). There is no explicit dependent service for `wpantund` (it uses DBus to provide API to other processes but that does not seem to be an issue here). 

`wantund` expects to get a socket path which it then uses to communicate to NCP 
- The socket can be to a serial device (UART) or another process, or input/output of an executable, or even an IP socket, etc). 
- When it starts, it tries to talk to NCP, by sending a RESET spinel command to NCP (over the socket path) and then waiting for a response from NCP.  
- If it times out and does not received a response, it will go through the same process again up to 3 attempts after which it stops and enters  "fault" state.

From the logs (and the fact that I think you mentioned you could not also connect to the board using CLI build?), my guess is that it is likely a config error, e.g., if the NCP build is set up to use UART (the same pins that device is connected to) with similar config (baud-rate, RTSCTS, etc).  Another thing to check is the FTDI driver? (but unlikely).

ChauhanPD, 2021-03-16:
Dbus status:
/usr/sbin# ps -ax | grep dbus
  250 ?        Ss     0:06 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
19054 ttymxc3  S+     0:00 grep dbus

I have disabled Hardware Flow control in OpenThread code of nrf52840 and flashed NCP hex file. Again verified baudrate, Pin numbers for Tx and Rx. Still the output of "wpantund" is same - retrying for 3 times and then State change: "uninitialized" -> "uninitialized:fault"

ChauhanPD, 2021-03-18:
Hello,

To remove any compile-time doubts, I have run the compilation in docker image of OpenThread (as mentioned in 1(b) of https://openthread.io/guides/build ).
Also, I have disabled Hardware Flow control in i.MX6 processor and in nrf52840 code. Baudrate is also given in wpantund command same as in nrf52840 code. Pin numbers are also verified many times.

Still, the error has not changed.

ChauhanPD, 2021-03-19:
I have flashed ot-cli-ftd firmware and opened that uart port on the terminal (using microcom) of i.MX6 processor. But there is no ">" and also no response for any command like state, reset.

Fortunately, I tried with one separate hardware module of nrf52840 which can be connected to our board via Mikrobus connector having another uart of i.MX6 host processor. I just changed the Tx / Rx pin number as per that hardware module in my code of nrf52840 which was used earlier.
I gave the same command of wpantund with other uart and wpantund is able to communicate with nrf52840.

Log is as below:
/usr/sbin/wpantund -o Config:NCP:SocketPath /dev/ttymxc2 -o Config:TUN:InterfaceName utun7 &
[1] 7595
root:~# wpantund[7595]: Starting wpantund 0.08.00d (Mar 11 2021 09:57:59) . . .
wpantund[7595]: 	SOURCE_VERSION = 0.07.01-354-ge8c110e
wpantund[7595]: Configuration file "/etc/wpantund.conf" read.
wpantund[7595]: Ready. Using DBUS bus ":1.14"
wpantund[7595]: Running as root without dropping privileges!
wpantund[7595]: State change: "uninitialized" -> "offline"
wpantund[7595]: NCP is running "OPENTHREAD/20191113-01806-g12d5f793e; NRF52840; Mar 18 2021 14:04:01"
wpantund[7595]: Driver is running "0.08.00d (0.07.01-354-ge8c110e; Mar 11 2021 09:57:59)"
wpantund[7595]: Network is not joinable
wpantund[7595]: [-NCP-]: SLAAC enabled
wpantund[7595]: Resetting interface(s). . .
wpantund[7595]: Finished initializing NCP

root:~# wpanctl -I utun7
wpanctl:utun7> status
utun7 => [
	"NCP:State" => "offline"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/20191113-01806-g12d5f793e; NRF52840; Mar 18 2021 14:04:01"
	"Daemon:Version" => "0.08.00d (0.07.01-354-ge8c110e; Mar 11 2021 09:57:59)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [F4CE36A0F93DD324]
]

Then tried to form a network and the log is as below:
wpanctl:utun7> form MyNetwork
Forming WPAN "MyNetwork" as node type "router"
wpantund[7595]: State change: "offline" -> "associating"
wpantund[7595]: ../../../git/src/ncp-spinel/SpinelNCPTaskForm.cpp:242: Check Failed (error 4)
wpantund[7595]: State change: "associating" -> "associated"
wpantund[7595]: Node type change: "unknown" -> "leader"
Successfully formed!

After some time again giving the command:
root:~# wpantund[31707]: Starting wpantund 0.08.00d (Mar 11 2021 09:57:59) . . .
wpantund[31707]:        SOURCE_VERSION = 0.07.01-354-ge8c110e
wpantund[31707]: Configuration file "/etc/wpantund.conf" read.
wpantund[31707]: Ready. Using DBUS bus ":1.30"
wpantund[31707]: Running as root without dropping privileges!
wpantund[31707]: State change: "uninitialized" -> "associating"
wpantund[31707]: ../../../git/src/ncp-spinel/SpinelNCPInstance-Protothreads.cpp:430: Requirement Failed (!ncp_state_is_joini)
wpantund[31707]: Resetting and trying again... (retry 1)
wpantund[31707]: State change: "associating" -> "uninitialized"
wpantund[31707]: [-NCP-]: NCP was reset (RESET_SOFTWARE, 114)
wpantund[31707]: State change: "uninitialized" -> "offline"
wpantund[31707]: NCP is running "OPENTHREAD/20191113-01806-g12d5f793e; NRF52840; Mar 18 2021 14:04:01"
wpantund[31707]: Driver is running "0.08.00d (0.07.01-354-ge8c110e; Mar 11 2021 09:57:59)"
wpantund[31707]: Network is not joinable
wpantund[31707]: State change: "offline" -> "offline:commissioned"
wpantund[31707]: [-NCP-]: SLAAC enabled
wpantund[31707]: Finished initializing NCP
wpantund[31707]: AutoResume is enabled. Trying to resume.
wpantund[31707]: NCP is commissioned. Resuming...
wpantund[31707]: State change: "offline:commissioned" -> "associating"
wpantund[31707]: State change: "associating" -> "associated"
wpantund[31707]: Node type change: "unknown" -> "leader"
wpantund[31707]: Adding multicast address "ff02::1:3" to NCP

root:~# wpanctl -I utun7
wpanctl:utun7> status
utun7 => [
        "NCP:State" => "associated"
        "Daemon:Enabled" => true
        "NCP:Version" => "OPENTHREAD/20191113-01806-g12d5f793e; NRF52840; Mar 18 2021 14:04:01"
        "Daemon:Version" => "0.08.00d (0.07.01-354-ge8c110e; Mar 11 2021 09:57:59)"
        "Config:NCP:DriverName" => "spinel"
        "NCP:HardwareAddress" => [F4CE36A0F93DD324]
        "NCP:Channel" => 26
        "Network:NodeType" => "leader"
        "Network:Name" => "MyNetwork"
        "Network:XPANID" => 0xEB40BB40E9568CA3
        "Network:PANID" => 0x1055
        "IPv6:LinkLocalAddress" => "fe80::98df:3fae:5dcc:6958"
        "IPv6:MeshLocalAddress" => "fdeb:40bb:40e9:0:d268:8c6e:c6ba:9fa9"
        "IPv6:MeshLocalPrefix" => "fdeb:40bb:40e9::/64"
        "com.nestlabs.internal:Network:AllowingJoin" => false
]
wpanctl:utun7> scan
   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
wpanctl:utun7>

So, can we say that everything is set on HOST side (OpenThread Boarder Router)?

Next, I should take one nrf52840 EVK as end device and join the network from there, right?

jwhui, 2021-03-19:
@ChauhanPD , yes it appears that wpantund and the NCP are communicating properly.

ChauhanPD, 2021-03-22:
@jwhui Thanks for confirming. I have taken nrf52840 EVK and ran CLI image on it. With that below are the logs

-----------
Host Board:
wpanctl:utun7> comissioner joiner-add f4ce3601f0a1258 100 "J01NME"
Added Joiner F4:CE:36:01:F0:A1:25:86, timeout:100, PSKd:"J01NME"
--------------
nrf52840 EVK:
>                                                                             
> joiner start J01NME                                                         
Done                                                                          
> Join success                                                                
                                                                              
> thread start
Done
> ifconfig
up
Done
> ping fdeb:40bb:40e9:0:d268:8c6e:c6ba:9fa9
Done
> 16 bytes from fdeb:40bb:40e9:0:d268:8c6e:c6ba:9fa9: icmp_seq=1 hlim=64 time=47ms

> state                                                                       
router                                                                        
Done  
-----------------------------------------

So, nrf52840 EVK has joined the network created by leader on Host processor and there is ping command response.

Next query is:
I would like to use RCP image (not NCP) for my nrf52840 connected with Host processor. So, in this case, what changes are required in ot-br-posix code because when nrf52840 is running with RCP image, nrf52840 is just a Radio gateway for OpenThread network communication and the Host processor will perform all the activities related to network communication.

ChauhanPD, 2021-03-22:
Also, with running my Host processor as leader and nrf52840 EVK as router, when I open web-page on my cellphone (running Android OS 10), I checked status and I found 
"WPAN service" as "uninitialized" and 
"wpantund" as "down"

What are the reasons for this?

I have started wpantund service from command line and when I run below command
systemctl start wpantund
it says 
the interface ttymxc2 is locked
which is obvious as wpantund is already running from as command (as shown in my post from 19th March).
Is this the reason for 
"wpantund" as "down"
on web-page?

ChauhanPD, 2021-04-05:
Hello All,

I was able to see everything correct on web-page on next reboot and did everything from web-page.

But as I have also asked earlier:
The next query is:
I would like to use RCP image (not NCP) for my nrf52840 connected with Host processor. So, in this case, what changes are required in ot-br-posix code? Because when nrf52840 is running with RCP image. nrf52840 is just a Radio gateway for OpenThread network communication and the Host processor will perform all the activities related to network communication. Or is it like that no change required in ot-br-posix code and it will do the required functionality in current code?

jwhui, 2021-04-05:
When using the RCP image, you should use the main branch of [openthread/ot-br-posix](https://github.com/openthread/ot-br-posix/).

We recently published a codelab with all the latest features - [Thread Border Router - Bidirectional IPv6 Connectivity and DNS-Based Service Discovery](https://openthread.io/codelabs/openthread-border-router#0). I encourage you to check it out.

ChauhanPD, 2021-04-06:
One observation:
When I program the hex file of NCP using nrfconnect 3.6.1 and read back from memory, it shows two areas in the rectangle area at left side - one is  green at top indicating "Application" as region name having 4 bytes and another is dark yellow at the bottom indicating "MBR or Application" as region name.

But when I program the hex file of RCP using same nrfconnect 3.6.1 and read back from memory, it shows only one area of dark yellow color indicating "MBR or Application".

Is this the reason, RCP image is not working - Host processor (running wpantund service) not able to communicate with nrf52840 having RCP image?

If yes, what is the solution?

Kindly guide or help me.

abtink, 2021-04-06:
@ChauhanPD 
- `wpantund` on its own requires an NCP to interact with.
- If you want to use RCP, you can follow https://github.com/openthread/ot-br-posix/

Let me also point you to https://openthread.io/platforms#platform_designs which provides an overview of how different architectures works.

ChauhanPD, 2021-04-06:
Thanks @abtink .

I have gone through  https://github.com/openthread/ot-br-posix/ . But haven't found changes required in wpantund or openthread or ot-br to work with RCP image.

I have gone through all pages of https://openthread.io/guides/border-router/build but nothing specific found for RCP image. With all those steps, I can successfully communicate with NCP image (running on nrf52840) from Host processor. I can also create (or form) a network and connect with router (having CLI application running on EVK of nrf52840).

But I can not communicate to the same nrf52840 having RCP image built from the same code of openthread.

There should be some change in openthread code side or change in wpantund. I have taken newer commit of wpantund, so, it should be compatible with both NCP and RCP..

Or am I missing something? 

Thanks.

abtink, 2021-04-06:
@ChauhanPD, `wpantund` works with an NCP. If you intend to use `wpantund` to directly interact with the RF chip running an RCP build, it won't work and was never supported (basically you would be missing the OpenThread protocol stack/implementation). 

`wpantund` needs to interact with an NCP. Now this can be realized  different ways:
- `wpantund` interacts directly with RF chip that is running NCP build (OpenThread stack is running on the chip).
- `wpantund` interacts with a `ot-ncp` posix process running host (OpenThread stack is running on the host in a process). 
   - In this case, the `ot-ncp` process itself interacts with RF chip running an RCP build (acting as a radio). 
   - Note that `ot-ncp` is executable output of OpenThread when built with "posix" platform (OpenThread POSIX app). You check the [README](https://github.com/openthread/openthread/tree/main/src/posix#readme) under posix.

jwhui, 2021-04-06:
@ChauhanPD , as I mentioned in https://github.com/openthread/openthread/issues/6283#issuecomment-813480702, the main branch of [openthread/ot-br-posix](https://github.com/openthread/ot-br-posix) supports RCP and does not include `wpantund`.

The `wpantund/master` branch of [openthread/ot-br-posix](https://github.com/openthread/ot-br-posix) supports NCP and does include `wpantund`.

ChauhanPD, 2021-04-08:
Hello,

I have updated the commit id in bb file of ot-br-posix and used below one:
##Main branch Commits till Apr 06, 2021
SRCREV = "b9565b4c44c9ccb413748d82905c8ef741768f57"

I have also updated the openthread source code with below commit id:
##Apr 7, 2021
OPEN_THREAD_COMMIT_HASH="83cecbb8e3f9418e3953e3ce65c269561b3f947b"

Using our .bb file I was able to compile the recipe for ot-br-posix after one modification to disable the otbr-web.service.

I have used the generated image and there was "ot-ctl" command available and no "wpanctl".
I have updated the /etc/default/otbr-agent file for correct path of communication (/dev/ttymxc2)
When I opened that file (/etc/default/otbr-agent), the line has
"-B wlan0" 
which I removed; as it was not mentioned in (https://openthread.io/guides/border-router/build )

but when I execute "ot-ctl state" or "ot-ctl reset" command, I observed error as below:

root@imx6ull:~# ot-ctl state
connect session failed: No such file or directory

root@imx6ull:~# ot-ctl reset
connect session failed: No such file or directory

Kindly guide me further.

ChauhanPD, 2021-04-09:
Hello All,

I got a discussion about my above issue related to ot-ctl at [this](https://github.com/openthread/ot-br-posix/issues/644).

But again, to compile an RCP image that can communicate with the Host successfully, which compiler flags must be set?

I have tried below,
make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 TMF_PROXY=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 DISABLE_SPI=1 SNTP_CLIENT=1 DISABLE_BUILTIN_MBEDTLS=1

but not working (same error: connect session failed: No such file or directory)
Also, after this, otbr-agent keeps restarting (because it cannot communicate with RCP, right?).

Thanks and Regards.

wgtdkp, 2021-04-09:
Hi,

When I opened that file (/etc/default/otbr-agent), the line has
> "-B wlan0"
> which I removed; as it was not mentioned in (
> https://openthread.io/guides/border-router/build )
>
I think you need the "-B wlan0" option.

Could you provide the output of "cat /var/log/syslog | grep otbr-agent" ?

BRs,
kangping

>
>

ChauhanPD, 2021-04-09:
Thanks, @wgtdkp for your quick reply.

I have restored that option (-b wlan0).
But observed the same error.

output of "cat /var/log/syslog | grep otbr-agent" ?
Apr  9 09:54:24 imx6ull otbr-agent[4629]: Running 0.3.0-b9565b4c-dirty
Apr  9 09:54:24 imx6ull otbr-agent[4629]: Thread version: 1.2.0
Apr  9 09:54:24 imx6ull otbr-agent[4629]: Thread interface: wpan0
Apr  9 09:54:24 imx6ull otbr-agent[4629]: Backbone interface: wlan0
Apr  9 09:54:26 imx6ull otbr-agent[4629]: [CRIT]-PLAT----: HandleRcpTimeout() at /usr/src/debug/otbr/git-r0/git/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2177: RadioSpinelNoResponse
Apr  9 09:54:31 imx6ull otbr-agent[4780]: Running 0.3.0-b9565b4c-dirty
Apr  9 09:54:31 imx6ull otbr-agent[4780]: Thread version: 1.2.0
Apr  9 09:54:31 imx6ull otbr-agent[4780]: Thread interface: wpan0
Apr  9 09:54:31 imx6ull otbr-agent[4780]: Backbone interface: wlan0
Apr  9 09:54:33 imx6ull otbr-agent[4780]: [CRIT]-PLAT----: HandleRcpTimeout() at /usr/src/debug/otbr/git-r0/git/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2177: RadioSpinelNoResponse
Apr  9 09:54:38 imx6ull otbr-agent[4927]: Running 0.3.0-b9565b4c-dirty
Apr  9 09:54:38 imx6ull otbr-agent[4927]: Thread version: 1.2.0
Apr  9 09:54:38 imx6ull otbr-agent[4927]: Thread interface: wpan0
Apr  9 09:54:38 imx6ull otbr-agent[4927]: Backbone interface: wlan0
Apr  9 09:54:40 imx6ull otbr-agent[4927]: [CRIT]-PLAT----: HandleRcpTimeout() at /usr/src/debug/otbr/git-r0/git/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2177: RadioSpinelNoResponse

wgtdkp, 2021-04-09:
It seems the communication between the otbr-agent and RCP is not working.

make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1
> DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 TMF_PROXY=1
> DHCP6_SERVER=1 DHCP6_CLIENT=1 DISABLE_SPI=1 SNTP_CLIENT=1
> DISABLE_BUILTIN_MBEDTLS=1
>
Looks like the very important USB or BOOTLOADER option is missing for the
nRF52xxx boards.

OpenThread is switching to maintain platform-specific examples in
separate repos, could you try to build the RCP with
the https://github.com/openthread/ot-nrf528xx repo and follow its guide
https://github.com/openthread/ot-nrf528xx/blob/main/README.md ?

BRs,
kangping

ChauhanPD, 2021-04-10:
Hello @wgtdkp 

Let me tell you that the nrf52840 chip to flash the RCP image is on a custom Plug-in Module on our custom board having an i.mx6ull processor. I am not using nrf52840 Dongle or EVK for the RCP image. So, I think in my case, the USB or BOOTLOADER option should not be used. Kindly correct me, if needed.

So, for my set-up, can I use  https://github.com/openthread/ot-nrf528xx repo?

> make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1

Same error.

wgtdkp, 2021-04-11:
> So, I think in my case, the USB or BOOTLOADER option should not be used. Kindly correct me, if needed.

I think you are right.

Can you provide the `border-agent` configuration file?

ChauhanPD, 2021-04-11:
root@imx6ull:~# vi /etc/default/otbr-agent 
# Default settings for otbr-agent. This file is sourced by systemd

# Options to pass to otbr-agent
OTBR_AGENT_OPTS="-I wpan0 -B wlan0 spinel+hdlc+uart:///dev/ttymxc2"
#OTBR_AGENT_OPTS="-I wpan0  spinel+hdlc+uart:///dev/ttymxc2"

ChauhanPD, 2021-04-12:
vi etc/cmake/options.cmake
 33 option(OTBR_BACKBONE_ROUTER "Enable Backbone Router" OFF)                       
 34 if (OTBR_BACKBONE_ROUTER)                                                       
 35     target_compile_definitions(otbr-config INTERFACE OTBR_ENABLE_BACKBONE_ROUTER=1)
 36 endif()                                                                         
 37                                                                                 
 38 option(OTBR_BORDER_ROUTING "Enable Border Routing Manager" OFF)                 
 39                                                                                 
 40 option(OTBR_DBUS "Enable DBus support" OFF)

Are line numbers 33 and 40 correct?

jwhui, 2021-04-12:
> vi etc/cmake/options.cmake
> 33 option(OTBR_BACKBONE_ROUTER "Enable Backbone Router" OFF)
> 34 if (OTBR_BACKBONE_ROUTER)
> 35 target_compile_definitions(otbr-config INTERFACE [...]

@ChauhanPD , these lines seem fine to me. These `option()` command declares an option for the user. Do you have any specific concern with these lines?

ChauhanPD, 2021-04-13:
option(OTBR_BACKBONE_ROUTER "Enable Backbone Router" OFF)

For Thread Version 1.2, OTBR_BACKBONE_ROUTER should be ON, right?

@wgtdkp below information is what you have asked for.
```
root@imx6ull:~# vi /etc/default/otbr-agent
#Default settings for otbr-agent. This file is sourced by systemd

#Options to pass to otbr-agent
OTBR_AGENT_OPTS="-I wpan0 -B wlan0 spinel+hdlc+uart:///dev/ttymxc2"
#OTBR_AGENT_OPTS="-I wpan0 spinel+hdlc+uart:///dev/ttymxc2"
```

jwhui, 2021-04-13:
> option(OTBR_BACKBONE_ROUTER "Enable Backbone Router" OFF)
> 
> For Thread Version 1.2, OTBR_BACKBONE_ROUTER should be ON, right?

Yes, this is set here:
https://github.com/openthread/ot-br-posix/blob/9242530cf66d6c6ae06d72400f17d9b711c91c0b/script/_otbr#L83

ChauhanPD, 2021-04-13:
Thanks, @jwhui .

Ok. So, OTBR_BACKBONE_ROUTER is made ON as below:
File: script/_otbr

if with BACKBONE_ROUTER; then
        otbr_options+=(
            "-DOTBR_BACKBONE_ROUTER=ON"
        )
    fi

And "BACKBONE_ROUTER" is set "1" as below:
examples/platforms/ubuntu/default:35:BACKBONE_ROUTER=1

But further query is "from where ubuntu/default is called?"

jwhui, 2021-04-13:
> But further query is "from where ubuntu/default is called?"

https://github.com/openthread/ot-br-posix/blob/75ef16d55e24c65bd5a89c9696e99591caf639c7/script/_initrc#L69

ChauhanPD, 2021-04-13:
Thanks, @jwhui 

script/bootstrap:34:. "$(dirname "$0")"/_initrc
So, in the case of the Yocto build system, I have to call 
`"script/bootstrap"`
from a recipe before compilation. Right?

jwhui, 2021-04-13:
> script/bootstrap:34:. "$(dirname "$0")"/_initrc
> So, in the case of the Yocto build system, I have to call
> `"script/bootstrap"`
> from a recipe before compilation. Right?

`script/bootstrap` is for installing dependencies.

Note that `script/_initrc` is also used by `script/setup`.

ChauhanPD, 2021-04-13:
yes. So, script/setup should be called from a recipe, right?

jwhui, 2021-04-13:
> yes. So, script/setup should be called from a recipe, right?
> But for NCP, I didn't have to do this. It was much easier. If you wish to have a look at my recipe used for NCP, I will share it.

It depends on your goals. `./script/setup` configures a number of other Linux-based services, including NAT64, DNS64, DHCPv6, network manager, etc. That script has always served this purpose, even in the [wpantund branch](https://github.com/openthread/ot-br-posix/tree/wpantund/master).

If you just want to build OpenThread, then you can figure out the CMake options you want and call CMake/Ninja directly.

ChauhanPD, 2021-04-13:
Thanks @jwhui for clarification.

My goals are simple:
- Create Thread Network with RCP image
- Send / Receive messages via the network.

So, Which flags should be ON and how to compile (like one way suggested to use CMake/Ninja directly)?

Thanks and Regards.

jwhui, 2021-04-13:
@ChauhanPD , I would suggest starting with either the Nordic nRF52840-DK or USB dongle and a Raspberry Pi as a reference. Once you get that working, you determine what you need to customize for your platform.

ChauhanPD, 2021-04-16:
Thanks @jwhui 

For that experiment also, I need to know which flags should be ON for RCP image to make it work with Host processor's ot-br-agent/ot-ctl.

jwhui, 2021-04-16:
> For that experiment also, I need to know which flags should be ON for RCP image to make it work with Host processor's ot-br-agent/ot-ctl.

See the [Setup OTBR](https://openthread.io/codelabs/openthread-border-router#1) section of the codelab.

ChauhanPD, 2021-04-16:
> USB_trans -DOT_THREAD_VERSION=1.2

I have tried "OT_THREAD_VERSION=1.2", but not "USB_trans" as I thought it would be for nrf52840 Dongle or EVK.
But still with "OT_THREAD_VERSION=1.2", I am not able to communicate from Host processor by running ot-ctl with nrf52840  having RCP image. Error I observed is same "connect session failed: No such file or directory".  Should I use "USB_trans" option?

To build RCP I follow below link and command:
https://openthread.io/guides/build
projects/git/openthread/openthread$ make -f examples/Makefile-nrf52840 clean && make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 OT_THREAD_VERSION=1.2 DAEMON=1
Is this okay?
My commit ID is: 83cecbb8e3f9418e3953e3ce65c269561b3f947b ( Wed Apr 7 02:09:09 2021 +0200 )

ChauhanPD, 2021-04-16:
my (yocto) compilation output is as below:

CMake Error at build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/otbr/git-r0/recipe-sysroot-native/usr/share/cmake-3.14/Modules/FindBoost.cmake:2132 (message):
|   Unable to find the requested Boost libraries.
| 
|   Boost version: 1.65.1
| 
|   Boost include path: /usr/include
| 
|   Could not find the following static Boost libraries:
| 
|           boost_filesystem
|           boost_system
| 
|   No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
|   directory containing Boost libraries or BOOST_ROOT to the location of
|   Boost.
| Call Stack (most recent call first):
|   etc/cmake/options.cmake:78 (find_package)
|   CMakeLists.txt:37 (include)
| 
| 
| -- *** (OTBR_VERSION) *** Version: 0.3.0-b9565b4c-dirty
| -- Checking for module 'systemd'
| -- List:  systemd
| -- Pkg:  systemd
| --   Found systemd, version 239
| -- Check if the system is big endian
| -- Searching 16 bit integer
| -- Looking for sys/types.h
| -- Looking for sys/types.h - found
| -- Looking for stdint.h
| -- Looking for stdint.h - found
| -- Looking for stddef.h
| -- Looking for stddef.h - found
| -- Check size of unsigned short
| -- Check size of unsigned short - done
| -- Using unsigned short
| -- Check if the system is big endian - little endian
| -- Version: 20191113-01878-g83cecbb8e-dirty
| -- OpenThread Config File: "openthread-core-posix-config.h"
| -- Found PythonInterp: build/tmp/hosttools/python (found version "2.7.17")
| -- Found Perl: build/tmp/hosttools/perl (found version "5.26.1")
| -- Performing Test FLAG_SUPPORTED_fvisibilityhidden
| -- Performing Test FLAG_SUPPORTED_fvisibilityhidden - Success
| -- Configuring incomplete, errors occurred!
| See also "build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/otbr/git-r0/git/build/otbr/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_config_extra (log file is located at build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/otbr/git-r0/temp/log.do_config_extra.2759)
ERROR: Task (sources/customer/meta-otbr/meta-otbr/recipes-core/otbr/otbr_git.bb:do_config_extra) failed with exit code '1

Anyone to guide/hint/help with the above error?

jwhui, 2021-04-16:
> I have tried "OT_THREAD_VERSION=1.2", but not "USB_trans" as I thought it would be for nrf52840 Dongle or EVK.
> But still with "OT_THREAD_VERSION=1.2", I am not able to communicate from Host [...]

The DK has two USB ports. If you connect to the nRF USB port, then you should us the `USB_trans` option. This is discussed in [the guide](https://openthread.io/codelabs/openthread-hardware#3) that I have linked a few times above.

> To build RCP I follow below link and command:
> https://openthread.io/guides/build
> projects/git/openthread/openthread$ make -f examples/Makefile-nrf52840 clean && make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 OT_THREAD_VERSION=1.2 DAEMON=1
> Is this okay?
> My commit ID is: [83cecbb](https://github.com/openthread/openthread/commit/83cecbb8e3f9418e3953e3ce65c269561b3f947b) ( Wed Apr 7 02:09:09 2021 +0200 )

You shouldn't need any options if you are using UART. `make -f examples/Makefile-nrf52840` should be sufficient in this case.

ChauhanPD, 2021-04-21:
Thanks @jwhui 

I will compile RCP image without any flag, but ot-br-posix compilation is having an error related to "Boost" folder. Is "Boost" compulsory for ot-br-posix compilation? If Yes, may you guide/hint to resolve that error?

Regards.

jwhui, 2021-04-22:
> I will compile RCP image without any flag, but ot-br-posix compilation is having an error related to "Boost" folder. Is "Boost" compulsory for ot-br-posix compilation? If Yes, may you guide/hint to resolve that error?

Boost dependencies are normally installed as part of script/bootstrap:
https://github.com/openthread/ot-br-posix/blob/a362436119261c7ea8f74a778b0d2a1236579b4f/script/bootstrap#L65

I believe Boost is only required by the web service feature. If you do not require the web service feature, then you can disable it by setting the `OTBR_WEB` CMake option to `OFF`.

ChauhanPD, 2021-04-26:
Hello @jwhui 
Thanks for your reply and support.

One more query:
script/setup modifies and/or adds files to the file-system along with compilation. Right? So, in my case of Yocto compilation, it actually modifies files of my PC and compiles code of the yocto build folder. 
Is it okay to set up after compilation? Is there any dependency of set-up before compilation? Can we do only compilation and then set-up? If yes, kindly guide me to do the only compilation.

Regards.

jwhui, 2021-04-27:
@ChauhanPD , your current issues all seem related to cross-compiling OTBR for i.MX6ULL. If so, can you close this issue and open a new issue in https://github.com/openthread/ot-br-posix/ ?

closed 2021-04-28 by ChauhanPD