openthread/openthread #6291: [github-actions] test_advertising_proxy.py fails intermittently

jwhui, 2021-03-16:
**Describe the bug**

tests/scripts/thread-cert/border_router/test_advertising_proxy.py fails intermittently.

**Console/log output**

- [logs_61185.zip](https://github.com/openthread/openthread/files/6151675/logs_61185.zip)
- [thread-border-router-results.zip](https://github.com/openthread/openthread/files/6151676/thread-border-router-results.zip)

wgtdkp, 2021-03-17:
@jwhui Looking into the log, I think it is not the `test_advertising_proxy.py` that is failing:

```txt
../../third_party/openthread/repo/src/core/border_router/routing_manager.cpp:339:49: error: no matching function for call to 'ot::BorderRouter::RoutingManager::IsValidOmrPrefix(ot::NetworkData::OnMeshPrefixConfig&)'
2021-03-16T18:33:32.3521052Z          if (!IsValidOmrPrefix(onMeshPrefixConfig))
```

Seems we made some mistakes when rebasing https://github.com/openthread/openthread/pull/6274 (The `IsValidOmrPrefix` is removed from the header file).

For GitHub Action workflow, we should fail at the `Build OTBR Docker` step when there is any error. This is a known issue @simonlingoogle.

jwhui, 2021-03-17:
@wgtdkp , thanks for catching that.

I captured another failure in `test_advertising_proxy.py`:

- [logs_61259.zip](https://github.com/openthread/openthread/files/6153749/logs_61259.zip)
- [thread-border-router-results.zip](https://github.com/openthread/openthread/files/6153753/thread-border-router-results.zip)

Thoughts?

wgtdkp, 2021-03-17:
```
Traceback (most recent call last):
2021-03-17T04:24:36.2436119Z   File "./tests/scripts/thread-cert/border_router/test_advertising_proxy.py", line 146, in test
2021-03-17T04:24:36.2437073Z     self.host_check_mdns_service(host, '2001::2')
2021-03-17T04:24:36.2437952Z   File "./tests/scripts/thread-cert/border_router/test_advertising_proxy.py", line 183, in host_check_mdns_service
2021-03-17T04:24:36.2439030Z     self.assertEqual(ipaddress.ip_address(service['addresses'][0]), ipaddress.ip_address(host_addr))
2021-03-17T04:24:36.2440055Z AssertionError: IPv6Address('2001::1') != IPv6Address('2001::2')
```

We are expecting that the Advertising Proxy will update the host address when client updates its IPv6 address:
```python
client.srp_client_set_host_address('2001::2')
self.simulator.go(2)

self.check_host_and_service(server, client, '2001::2')
self.host_check_mdns_service(host, '2001::2')
```

The `dns-sd` log shows that mDNSResponder doesn't remove the stale AAAA record immediately when the record is updated:
```
cat /tmp/my-host
2021-03-17T04:24:36.1612953Z 2021-03-17 04:24:16,160 - INFO - Host<3> $ DATE: ---Wed 17 Mar 2021---
2021-03-17T04:24:36.1613714Z 2021-03-17 04:24:16,161 - INFO - Host<3> $  4:24:13.648  ...STARTING...
2021-03-17T04:24:36.1614761Z 2021-03-17 04:24:16,161 - INFO - Host<3> $ Timestamp     A/R Flags if Hostname                               Address                                      TTL
2021-03-17T04:24:36.1615648Z 2021-03-17 04:24:16,161 - INFO - Host<3> $  4:24:13.648  Add     3 30 my-host.local.                         2001:0000:0000:0000:0000:0000:0000:0001%<0>  120
2021-03-17T04:24:36.1616468Z 2021-03-17 04:24:16,161 - INFO - Host<3> $  4:24:13.648  Add     2 30 my-host.local.                         2001:0000:0000:0000:0000:0000:0000:0002%<0>  120
2021-03-17T04:24:36.1617637Z 2021-03-17 04:24:16,161 - INFO - Host<3> $  4:24:15.511  Rmv     0 30 my-host.local.                         2001:0000:0000:0000:0000:0000:0000:0001%<0>  0
```
The old address `2001::1` is removed 2 seconds after adding the new address `2001::2`.

I think wait a bit longer after updating the address should help, I will try at my local to see if it works.