openthread/openthread #6292: Wait For Response times out during RadioSpinel::Init  with aResetRadio = false

paragdixit-g, 2021-03-17:
We see that during RadioSpinel::Init with aResetRadio = false, the Openthread times-out waiting for response. 

This is because, WaitForResponse is done outside the conditional : 

==
    if (aResetRadio)
    {
        SuccessOrExit(error = SendReset());
    }

    SuccessOrExit(error = WaitResponse());
==

Is this is a bug ?  ie  WaitForResponse should be done within conditional ? 
Or Is the expectation that this function is called with  'aResetRadio = false'  only if rcp is reset earlier or is otherwise expected to send some response ?

abtink, 2021-03-17:
I think @bukepo may remember more on this and can help shed more light on this. 

But I don't think it's is a bug and it is actually intentional. 

As you said, the idea is that one way or other we expect the RCP to be reset before `Init()` is called. I think reset is required to ensure that RCP starts fresh with no previous state.
-  It can be done through a software command (sending an spinel RESET command to the RCP) which is done by `SendReset()` and controlled through `aResetRadio` param.
- Or it may be reset by the outside process which is starting RCP (and OT posix app). For example, we may have a hardware reset pin on RCP that is then used to explicit reset it.

> or is otherwise expected to send some response ?

We do need to see a spinel `RESET` event from RCP. Note that we have this block in `Init()`:
```
    SuccessOrExit(error = WaitResponse());
    VerifyOrExit(mIsReady, error = OT_ERROR_FAILED);
```
and `mIsReady` is set when a `RESET` spinel `LAST_STATUS` is received.

paragdixit-g, 2021-03-19:
Thanks for the explanation ! 

It may help to add to documentation of Init the expectation that RCP should be in 'init state' prior to calling the function with aResetRadio = false.

closed 2021-03-19 by paragdixit-g