openthread/openthread #6300: state=detached for RCP Joiner in codelab tutorial

squishykid, 2021-03-17:
**Describe the bug** When running through the guide, when it comes time to join the RCP Joiner to the network, I get the following error:
```
> dataset
Master Key: <key>
Done
> thread start
Done
> state
detached
Done
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: 12d5f793e7a99b34ca9a52a71d9d6df0207f2dd7
2. IEEE 802.15.4 hardware platform: nRF52840-DK
3. Follow guide to: https://openthread.io/codelabs/openthread-hardware#7 , then to the 4th pane from the top.

**Expected behavior** I expect `state` to be `child`

**Extra Info** I was able to flash the RCP Joiner hardware like the FTD Joiner and follow on with the instructions with two FTD joiners

squishykid, 2021-03-17:
The rest of the guide was great though, cool to see the UDP packets flying around the mesh

Irving-cl, 2021-03-17:
@squishykid 

How long did you wait after `thread start`? It may take a while (though not very long) for the device to attach. Does the RCP Joiner keep `detached` state?

squishykid, 2021-03-18:
Hi, apologies for the noise. After running through the guide again and waiting a bit longer it worked as described.

closed 2021-03-18 by squishykid