openthread/openthread #6310: nRF52840 dongle FW build broken

mikhail-pridushchenko, 2021-03-18:
**Describe the bug**
nRF52840 dongle firmware build broken for OTBR commit cf3a67383e9e25ca89208234d8efd8bee6b92f25 (merged PR 721)

**To Reproduce**

1. Git commit id 818a774bcf2cc655ff3d400f50c5b50d96d3e771
2. IEEE 802.15.4 hardware platform nRF52840 dongle
3. Build steps
```
$ make -f examples/Makefile-nrf52840 USB=1 BOOTLOADER=1 THREAD_VERSION=1.2
```

**Expected behavior**
No errors during build.

**Console/log output**
```
$ make -f examples/Makefile-nrf52840 USB=1 BOOTLOADER=1 THREAD_VERSION=1.2
...
  CC       src/transport/libopenthread_nrf52840_transport_a-usb-cdc-uart.o
  CC       src/transport/libopenthread_nrf52840_transport_a-transport.o
  AR       libopenthread-nrf52840-transport.a
make[5]: Nothing to be done for 'all-am'.
Making all in apps
Making all in cli
  CC       ot_cli_ftd-main.o
  CCLD     ot-cli-ftd
/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: ../../../src/cli/libopenthread-cli-ftd.a(libopenthread_cli_ftd_a-cli.o): in function `ot::Cli::Interpreter::OutputResult(otError)':
/home/otc/src/non-CCM/openthread/examples/../src/cli/cli.cpp:171: undefined reference to `otThreadErrorToString'
collect2: error: ld returned 1 exit status
Makefile:1717: recipe for target 'ot-cli-ftd' failed
make[5]: *** [ot-cli-ftd] Error 1
Makefile:429: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
Makefile:430: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:586: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:512: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-nrf52840:329: recipe for target 'do-build-cortex-m4' failed
make: *** [do-build-cortex-m4] Error 2
```

wgtdkp, 2021-03-18:
@mikhail-pridushchenko It looks like you didn't clear before building the new firmware, try:
```shell
rm -rf build output
./bootstrap
make -f examples/Makefile-nrf52840 USB=1 BOOTLOADER=1 THREAD_VERSION=1.2
```

mikhail-pridushchenko, 2021-03-18:
@wgtdkp Thank you.
`./bootstrap` step was missing. Pity is that no obvious error is produced in this case

closed 2021-03-18 by mikhail-pridushchenko