openthread/openthread #6312: [github-actions] test_dua_dad.py fails intermittently

jwhui, 2021-03-18:
**Describe the bug**

tests/scripts/thread-cert/backbone/test_dua_dad.py fails intermittently

**Console/log output**

- [logs_61772.zip](https://github.com/openthread/openthread/files/6165151/logs_61772.zip)
- [thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6165152/thread-1-2-backbone-results.zip)

jwhui, 2021-05-21:
[logs_79079.zip](https://github.com/openthread/openthread/files/6524617/logs_79079.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6524619/thread-1-2-backbone-results.zip)

jwhui, 2021-06-24:
[logs_83059.zip](https://github.com/openthread/openthread/files/6711653/logs_83059.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6711654/thread-1-2-backbone-results.zip)

simonlingoogle, 2021-06-25:
Got error:
```
2021-06-24T17:41:51.6591916Z ----------------------------------------------------------------------
2021-06-24T17:41:51.6592392Z Traceback (most recent call last):
2021-06-24T17:41:51.6593225Z   File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 282, in tearDown
2021-06-24T17:41:51.6593889Z     self._verify_packets(test_info_path)
2021-06-24T17:41:51.6594748Z   File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/thread_cert.py", line 310, in _verify_packets
2021-06-24T17:41:51.6595379Z     self.verify(pv)
2021-06-24T17:41:51.6596033Z   File "./tests/scripts/thread-cert/backbone/test_dua_dad.py", line 222, in verify
2021-06-24T17:41:51.6597909Z     pv.verify_dua_registration(ROUTER1, ROUTER1_DUA, pbbr_eth=PBBR_ETH, pbbr_src64=PBBR, sbbr_eth=SBBR_ETH)
2021-06-24T17:41:51.6599124Z   File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/pktverify/packet_verifier.py", line 332, in verify_dua_registration
2021-06-24T17:41:51.6600338Z     assert bq1.sniff_timestamp - dr.sniff_timestamp <= 1.0, bq1.sniff_timestamp - dr.sniff_timestamp
2021-06-24T17:41:51.6600936Z AssertionError: 1.0004706382751465
2021-06-24T17:41:51.6601187Z 
```

PR #6765 (2021-06-25): [dua] fix `test_dua_dad` random fails
Fixes #6312 

See https://github.com/openthread/openthread/issues/6312#issuecomment-868113264 for the cause.
closed 2021-06-25 by jwhui