openthread/openthread #6314: [github-actions] border router tests failing

jwhui, 2021-03-19:
**Console/log output**

- https://github.com/openthread/openthread/actions/runs/666906173
- https://github.com/openthread/openthread/actions/runs/666900234

wgtdkp, 2021-03-19:
The `Build OTBR Docker` step is failing:
```
FAILED: src/agent/CMakeFiles/otbr-agent.dir/ncp_openthread.cpp.o 
/usr/bin/c++  -DOPENTHREAD_SPINEL_CONFIG_OPENTHREAD_MESSAGE_ENABLE=0 -DOTBR_ENABLE_BACKBONE_ROUTER=1 -DOTBR_ENABLE_DBUS_SERVER=1 -DOTBR_ENABLE_DUA_ROUTING=1 -DOTBR_ENABLE_MDNS_MDNSSD=1 -DOTBR_ENABLE_SRP_ADVERTISING_PROXY=1 -DOTBR_PACKAGE_NAME=\"OPENTHREAD_BR\" -DOTBR_PACKAGE_VERSION=\"0.3.0\" -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I../../include -I../../src -Ithird_party/openthread/repo/etc/cmake -I../../third_party/openthread/repo/etc/cmake -I../../third_party/openthread/repo/include -I../../third_party/openthread/repo/src/posix/platform/include -I../../third_party/openthread/repo/src -Wall -Wextra -Werror -Wfatal-errors -Wno-missing-braces -std=c++11 -MD -MT src/agent/CMakeFiles/otbr-agent.dir/ncp_openthread.cpp.o -MF src/agent/CMakeFiles/otbr-agent.dir/ncp_openthread.cpp.o.d -o src/agent/CMakeFiles/otbr-agent.dir/ncp_openthread.cpp.o -c ../../src/agent/ncp_openthread.cpp
../../src/agent/ncp_openthread.cpp: In member function 'otbrError otbr::Ncp::ControllerOpenThread::Init()':
../../src/agent/ncp_openthread.cpp:110:5: error: 'otCliUartInit' was not declared in this scope
     otCliUartInit(mInstance);
     ^~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
```

jwhui, 2021-03-19:
@wgtdkp , yes, we are waiting on https://github.com/openthread/ot-br-posix/pull/722, but that is experiencing CI failures as well.

jwhui, 2021-03-19:
Resolved by https://github.com/openthread/ot-br-posix/pull/722

closed 2021-03-19 by jwhui