openthread/openthread #6319: How to simulate multiple RCP nodes?

purvisb, 2021-03-20:
[I've gone through this codelab](https://openthread.io/codelabs/openthread-simulation-posix#1)

I'd like to simulate multiple RCP nodes. How do I do this? Can I create multiple ot-daemon instances?

jwhui, 2021-03-20:
`ot-daemon` currently only supports a single instance. If you really want to run multiple instances with a simulated RCP, you can use multiple Docker container instances - see our guide for how to [attach a simulated RCP to a Docker container](https://openthread.io/guides/border-router/docker/run#simulated_rcp).

You can also run `ot-cli` with a simulated RCP - see [src/posix/README.md](https://github.com/openthread/openthread/blob/main/src/posix/README.md#with-simulation). However, that won't give you a network interface in Linux - just simulated ot-cli devices.

jwhui, 2021-03-30:
Closing stale issue.

closed 2021-03-30 by jwhui