openthread/openthread #6321: OT-Daemon Loses Off-mesh Prefix on Restart

gabekassel, 2021-03-20:
**Describe the bug**
ot-daemon will not preserve prefixes in network data for that host/router. Not clear if this is by design, but means prefixes need to be re-added manually if ot-daemon crashes and is restarted by a system manager (e.g. systemd)

**To Reproduce** Information to reproduce the behavior, including:
Start ot-daemon, add a prefix for that host, restart ot-daemon, ifconfig up/thread start. Prefix is no longer in netdata (even after upgrading back to router). 
```
root@eero:~# ot-ctl netdata show
Prefixes:
fd97:739d:386a:2::/64 paros med f800
Routes:
Services:
Done
root@eero:~# systemctl stop ot-daemon
root@eero:~# systemctl start ot-daemon
root@eero:~# ot-ctl state
router
Done
root@eero:~# ot-ctl netdata show
Prefixes:
Routes:
Services:
Done
root@eero:~# ot-ctl rloc16
f800
Done
```
1. Git commit id - 7b3c2ae
2. IEEE 802.15.4 hardware platform - Linux host + Silabs RCP
3. Build steps
4. Network topology - 1 BR + 1 WSTK REED

**Expected behavior** A clear and concise description of what you expected to happen.
Ideally the prefix would be retained/readded to netdata after the router comes back, however not sure if this is intentionally avoided. 

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2021-03-20:
@gabekassel  , are you using the [`OPENTHREAD_CONFIG_BORDER_ROUTING_ENABLE `](https://github.com/openthread/openthread/blob/de3ddb7169c85f8b1f1bf6fe322cd997d59dd1ba/src/core/config/border_router.h#L65) feature introduced in #5856? This feature should locally generate a prefix for use with off-mesh communication and store it in settings (see [src/core/border_router/routing_manager.cpp](https://github.com/openthread/openthread/blob/de3ddb7169c85f8b1f1bf6fe322cd997d59dd1ba/src/core/border_router/routing_manager.cpp#L123)).

As for other configurations, we intentionally try to minimize persistent state in OpenThread. If there are other prefixes/routes that you'd like to configure, we have so far left that up to the application to ensure those are re-added on a restart.

gabekassel, 2021-03-21:
@jwhui i'm not using that feature, but instead applying my own OMR from an out of band configuration. out of curiosity, if an OMR is provided (like I'm doing) with the mentioned feature, will it converge on my preferred prefix or have already set a new/random one? 

since we usually have 2+ BRs in a home, it made sense to try to have a pre-determined OMR with prefixes/routes for each BR in netdata.

jwhui, 2021-03-21:
@gabekassel , I think we'll need to think more about this. Currently, we are following the behavior in [Section 5 of draft-lemon-stub-networks-02](https://datatracker.ietf.org/doc/html/draft-lemon-stub-networks-02#section-5), which currently states:

```
   When the time comes to deprecate one or more prefixes as a result of
   a network partition healing, only one prefix should remain.  If there
   are any GUA prefixes, and if there is no specific configuration
   contradicting this, the GUA prefix that is numerically lowest should
   be kept, and all others deprecated.  If there are no GUA prefixes,
   then the ULA prefix that is numerically lowest should be kept, and
   the others deprecated.  By using this approach, it is not necessary
   for the routers to coordinate in advance.
```

In summary, if the randomly chosen OMR prefix is numerically lower than another one that is manually configured, the randomly chosen OMR prefix will still remain.

In your setup, does each BR have their own OMR prefix? Or do all the BRs share the same OMR prefix? If they all have their own OMR prefix, we could consider a way to explicitly configure the OMR prefix rather than randomly generating one. Basically, a way to configure the OMR prefix that is stored in settings.

gabekassel, 2021-03-21:
In our case currently, all BRs have the same OMR applied during bringup/merging/joining. 

I'm not sure how much value the custom OMR has, but it's derived as the second /64 from a /48 given to the home (first /64 goes to WiFi).

jwhui, 2021-03-21:
Applying the same OMR won't work if the Thread network ever partitions. But maybe you are depending on Thread-over-Infrastructure to just work :) Is there ever a case in your scenario where both Wi-Fi and Thread can be partitioned?

Do your configuration have a notion of "leader" or "root" node for Wi-Fi? Another option is that only the elected "root" serves as a border router, but that would mean all Wi-Fi-Thread traffic must traverse the single border router.

gabekassel, 2021-03-21:
> Applying the same OMR won't work if the Thread network ever partitions. But maybe you are depending on Thread-over-Infrastructure to just work :) Is there ever a case in your scenario where both Wi-Fi and Thread can be partitioned?
It's certainly possible wifi or thread couple independently partition, but feels unlikely both would partition at the same time. There are also transient cases where one BR is rebooting while others are still serving wifi/thread. 

> Do your configuration have a notion of "leader" or "root" node for Wi-Fi? Another option is that only the elected "root" serves as a border router, but that would mean all Wi-Fi-Thread traffic must traverse the single border router.
Yes we do have this concept when the network is in NAT mode. It felt very advantageous to leverage every available BR, so I've been trying to avoid designs like this.

Feels like TREL may resolve the partitioning concern, but perhaps enabling the border routing feature and letting the OMR be chosen at random is better?

jwhui, 2021-03-21:
> Feels like TREL may resolve the partitioning concern, but perhaps enabling the border routing feature and letting the OMR be chosen at random is better?

Standardizing on a single solution (i.e. `draft-lemon-stub-networks`) will likely be simpler to deal with, at least in the beginning. If we run into compelling reasons to support "out-of-band" configurations, then we can address those at that time.