openthread/openthread #6337: Throughput degradation with Thread 1.2 features enabled

KaSroka, 2021-03-23:
**Describe the bug**
During testing we have found out that enabling Thread 1.2 features results in decreased throughput in child to child communication.

**To Reproduce**
1. Flash the devkits with image build from latest main branch with Thread 1.2 features enabled. The devkitss are up and running.
2. Form topology: end-device <> router <> end-device (end devices are FTD, using NCP wpantund interface,
router uses CLI interface.
3. On the first devkit host, run the Iperf server, which will receive throughput from client.
Example UDP server command:
$ iperf -V -u -s -p 5000 -i 1
5. On the second devkit host, run the Iperf client, which will be sending throughput to server.
Example UDP client command:
$ iperf -V -u -c <IPv6 of DK1> -p 5000 -b 200k -l 1128 -i 1 -t 60
NOTE: use IPv6 mesh local
6. Verify that the server successfully receives data from the client.
7. Verify that the value of average throughput with Thread 1.2 features is ~3kbit/s.

**Expected behavior**
Throughput in point 7. with image build from same commit but with Thread 1.2 features disabled is ~45kbit/s

**Additional context**
- We are using nRF52840 devkits in this test but it looks like the problem is platform independent - I assume problem lays on MAC layer.
- We are using NCP + wpandund for end devices as this allows us to run iperf easily.
- We are not observing this problem in router <> router communication.

Can someone help with further investigation?

simonlingoogle, 2021-03-24:
Hi @KaSroka, 

Thanks for catching this issue. 

Could you sniffer the traffic and upload the PCAPs (both 1.1 and 1.2). This would be very helpful for identifying the cause.

luga-ns, 2021-03-24:
Thread 1.1:
build with:
make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COAPS=1 COMMISSIONER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 DNS_CLIENT=1 ECDSA=1 IP6_FRAGM=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 SERVICE=1 SLAAC=1 SNTP_CLIENT=1 UDP_FORWARD=1 THREAD_VERSION=1.1
results:
test_app_layer_thput_performance_short_child_router_child result comment: {'min': 36.1, 'max': 54.1, 'avg': 50.62, 'ninety_perc': 54.1, 'unit': 'k'}
[thread_1_1.zip](https://github.com/openthread/openthread/files/6196156/thread_1_1.zip)

Thread 1.2:
build with:
make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COAPS=1 COMMISSIONER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 DNS_CLIENT=1 ECDSA=1 IP6_FRAGM=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 SERVICE=1 SLAAC=1 SNTP_CLIENT=1 UDP_FORWARD=1 BACKBONE_ROUTER=1 CSL_RECEIVER=1 LINK_METRICS=1 DUA=1 MLR=1
results:
test_app_layer_thput_performance_short_child_router_child result comment: {'min': 0.0, 'max': 18.0, 'avg': 3.7, 'ninety_perc': 9.02, 'unit': 'k'}
[thread_1_2.zip](https://github.com/openthread/openthread/files/6196155/thread_1_2.zip)

pcaps and wpantund logs in attachments.

simonlingoogle, 2021-03-25:
> `Form topology: end-device <> router <> end-device`

Comparing the `thread_1_1` and `thread_1_2` PCAPs,  one difference is that in `thread_1_2`, the router (leader) do not forward the 1128 sized UDP datagrams as many as in `thread_1_2`.

For example, in the traffic shown below, the router only forwards two UDP datagrams, in the meantime the end device has sent tens of datagrams. 

[image]

Not sure why the router failed to forward many datagrams. One possible cause is that the router is running out of message buffers.
I would suggest to debug the issue on router with more logging.

luga-ns, 2021-03-25:
[Thread_1_2_debug_logs.zip](https://github.com/openthread/openthread/files/6204979/Thread_1_2_debug_logs.zip)
build with:
make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COAPS=1 COMMISSIONER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 DNS_CLIENT=1 ECDSA=1 IP6_FRAGM=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 SERVICE=1 SLAAC=1 SNTP_CLIENT=1 UDP_FORWARD=1 BACKBONE_ROUTER=1 CSL_RECEIVER=1 LINK_METRICS=1 DUA=1 MLR=1 FULL_LOGS=1 LOG_OUTPUT=APP
test result:
test_app_layer_thput_performance_short_child_router_child result comment: {'min': 0.0, 'max': 36.1, 'avg': 16.88, 'ninety_perc': 27.1, 'unit': 'k'}

result on build with debug logs is much better that without debug logs which is very strange for me, I'd expect even lower throughput with debug logs which we observe in scenario router - router.

simonlingoogle, 2021-03-25:
There seems to be many 6lowpan fragment dropping like below:

```
wpantund[183]: NCP => [0000000094278][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:NoBufs, len:100, src:0x3001, dst:0x3000, tag:8344, offset:0, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000094317][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8344, offset:216, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000094355][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8344, offset:408, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000094403][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8344, offset:600, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000094464][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8344, offset:888, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000094502][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8344, offset:1080, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097639][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:NoBufs, len:100, src:0x3001, dst:0x3000, tag:8355, offset:0, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097674][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8355, offset:120, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097713][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8355, offset:312, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097754][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8355, offset:504, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097786][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8355, offset:600, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097827][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8355, offset:792, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097861][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8355, offset:888, dglen:1176, sec:yes
wpantund[183]: NCP => [0000000097934][NOTE]-MAC-----: [NOTE]-MAC-----: Dropping rx frag frame, error:Drop, len:96, src:0x3001, dst:0x3000, tag:8355, offset:1080, dglen:1176, sec:yes
```

This happens when OT is running out of message buffers. 
It's unclear why Thread 1.2 has less available message buffers. I think we should check if:
- Thread 1.2 is allocating less number of message buffers overall
- What components have been allocating message buffers, and whether they have released the buffers.

luga-ns, 2021-03-26:
I've tested this also on Thread 1.1
[Thread_1_1_logs.zip](https://github.com/openthread/openthread/files/6210847/Thread_1_1_logs.zip)
the same logs with error:NoBufs and error:Drop in Thread 1.1.
Result in case of build with debug logs is almost the same as for Thread_1_2:
test_app_layer_thput_performance_short_child_router_child result comment: {'min': 0.0, 'max': 27.1, 'avg': 16.72, 'ninety_perc': 27.1, 'unit': 'k'}
(problem only exists on build without flags FULL_LOGS=1 LOG_OUTPUT=APP ?)

In this scenario I try to send 130kbps which is more than radio can handle so we expect that some packet will be dropped. 
I've tested the same scenario with Thread 1.2 but now iperf tried to send only 45kbps and then this scenario works as expected. 
Also results are much better when I try to sent 130kbps but packets are shorter, e.g. 100 instead of 1128.

Just a guess but I think that it might be some issue with dropping packets/fragments in case of Thread 1.2, some fragments are dropped but remaining fragments from the same packet are still send (and lost fragments are not correctly retransmitted), then receiver reports low throughput as it didn't receive all fragments from the same packet.

jwhui, 2021-04-27:
Resolved by #6458

closed 2021-04-27 by jwhui