openthread/openthread #6338: how to setup OTBR with wpantund to connect with a masterkey

cahty82008, 2021-03-23:
I followed this tutorial with Step 7 and Step 8  to set up a OTBR and join it with the masterkey 
[https://openthread.io/codelabs/openthread-hardware#6](url)

I set up a border router with wpanctl(the tutorial is using a CLI example), but i have problem to connect to this border router. Are there any other parameter that i should set up on OTBR? 

**set up the OTBR:**
```
wpanctl leave                                                     
wpanctl reset                                                                                                               
wpanctl setprop Dataset:MasterKey 00112233445566778899aabbccddffee
wpanctl form "OpenThreadTest"
```
**set up a joiner with CLI example:**
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
> | 0 | OpenThreadTest   | 39741e55e59e5ab4 | de1a | 4ead76f01a2edac3 | 19 | -43 | 200 |
> dataset masterkey 00112233445566778899aabbccddffee
Done
> dataset commit active
Done
> dataset
Master Key: 00112233445566778899aabbccddffee
Done
> thread start
Done
> state
detached
Done

jwhui, 2021-03-24:
@cahty82008 , it appears that the master keys do not match:
```
wpanctl setprop Network:Key 11112233445566778899DEAD1111DEAD
```
vs.
```
dataset masterkey 00112233445566778899aabbccddffee
```

cahty82008, 2021-03-25:
> @cahty82008 , it appears that the master keys do not match:
> 
> ```
> wpanctl setprop Network:Key 11112233445566778899DEAD1111DEAD
> ```
> 
> vs.
> 
> ```
> dataset masterkey 00112233445566778899aabbccddffee
> ```

 thanks for the reply. At first I thought 
`dataset masterkey 00112233445566778899aabbccddffee`
should match with 
`wpanctl setprop Dataset:MasterKey 00112233445566778899aabbccddffee`

i changed dataset masterkey from '00112233445566778899aabbccddffee' to '11112233445566778899DEAD1111DEAD', and i also runed `wpanctl permit-join 3600 1212` 

but still i can not connect to my OTBR

status of joiner:
`> dataset
Master Key: 11112233445566778899dead1111dead
Done
> state
detached
Done
> ipaddr
fdde:ad00:beef:0:273d:4ee2:3874:6850
fe80:0:0:0:64d6:86b6:efe8:db04
Done`

status of OTBR:
`# wpanctl status
wpan0 => [
	"NCP:State" => "associated"
	"Daemon:Enabled" => true
	"NCP:Version" => "OPENTHREAD/20191113-00534-gc6a258e3; NRF52840; Apr  5 2020 21:51:18"
	"Daemon:Version" => "0.08.00d (; Apr 21 2020 19:11:43)"
	"Config:NCP:DriverName" => "spinel"
	"NCP:HardwareAddress" => [F4CE3699222F6E93]
	"NCP:Channel" => 25
	"Network:NodeType" => "leader"
	"Network:Name" => "OpenThreadTest"
	"Network:XPANID" => 0xDEAD1111DEAD2222
	"Network:PANID" => 0xDEAD
	"IPv6:LinkLocalAddress" => "fe80::141f:xxxx:xxxx:xxxx"
	"IPv6:MeshLocalAddress" => "fdde:ad11:11de:0:8fce:xxxx:xxxx:xxxx"
	"IPv6:MeshLocalPrefix" => "fdde:ad11:11de::/64"
	"com.nestlabs.internal:Network:AllowingJoin" => true
]
`

jwhui, 2021-03-25:
@cahty82008 , I believe your master key is still not matching:
```
wpanctl setprop Dataset:MasterKey 00112233445566778899aabbccddffee
```
vs.
```
Master Key: 11112233445566778899dead1111dead
```

cahty82008, 2021-03-25:
@jwhui   hi jwhui, please notice that i in my OTBR **i set up two keys:**
**OTBR:**
`wpanctl setprop Network:Key 11112233445566778899DEAD1111DEAD
wpanctl setprop Dataset:MasterKey 00112233445566778899aabbccddffee`

**and in the joiner, i tried both keys as the masterkey, unfortunately both of them are not working:**
**joiner first try:**
```
> factoryreset
> ifconfig up
Done
> dataset masterkey 11112233445566778899DEAD1111DEAD
Done
> dataset
Master Key: 11112233445566778899dead1111dead
Done
> dataset commit active
Done
> thread start
Done
> state
detached
Done
```
**joiner second try:**
```
> factoryreset
> ifconfig up
Done
> dataset masterkey 00112233445566778899aabbccddffee
Done
> dataset
Master Key: 00112233445566778899aabbccddffee
Done
>  dataset commit active
Done
> thread start
Done
> 
> state
detached
Done
```

jwhui, 2021-04-06:
@cahty82008 , with wpanctl, just try setting the masterkey one way:

For example:
```
wpanctl leave
wpanctl setprop Network:PANID 0xabcd
wpanctl setprop Network:XPANID dead00beef00cafe
wpanctl setprop Network:Key 00112233445566778899aabbccddeeff
wpanctl form -c 11 "OpenThread"
```

jwhui, 2021-04-26:
Closing stale issue.

closed 2021-04-26 by jwhui