openthread/openthread #6344: [posix] ot-br-posix dry run will be affected by failed initialization of irrelavant modules

Irving-cl, 2021-03-24:
**Describe the bug** A clear and concise description of what the bug is.
When I run ot-br-posix on a device, I found that I cannot correctly get RCP version by using `--radio-version`. The reason is that `platformBackboneInit` in `otSysInit` failed and the program directly exited. The code to handle `radio-version` is after `otSysInit` and doesn't have a chance to run. I think it's possible that some devices don't have backbone interface or temporarily the interface doesn't work. But it should break other dry-runs of ot-br-posix. For example, users just want to output the RCP version. Then it should be correctly output even backbone is missing.

I cannot think of a 'perfect' implementation. My idea is that we keep the declaration of `otSysInit` for posix, and we devide it into two functions: one before `Instance` created, one after `Instance` created. And we put those initialization that are dependent on `Instance` (e.g. `platformBackboneInit`) in the second function. And `InitInstance` looks like:
```
instance = otSysInit(&aConfig->mPlatformConfig);

atexit(otSysDeinit);

if (aConfig->mPrintRadioVersion)
{
    ......
}
else
{
    ......
}

if (aConfig->mIsDryRun)                                                                                                                                                                                                                                                                                     
{                                                                                                                                                                                                                                                                                                           
    ......                                                                                                                                                                                                                                                  
}

otSysInitWithInstance(&aConfig->mPlatformConfig, instance); // do things like `platformBackboneInit`
```

I think when we do dry-run, we don't actually need to initialize everything. Thoughts? @bukepo @librasungirl @wgtdkp @abtink

bukepo, 2021-03-25:
dry run can be used to test the stack can be initialized successfully.

simonlingoogle, 2021-03-29:
Is this feature absolute required in some scenario ? I think we can just specify `-B ....` even for dry runs. It seems a overkill for me to implement such feature just to type less letters.

closed 2021-03-29 by Irving-cl