openthread/openthread #6349: [github-actions] test_srp_register_single_service.py fails intermittently

jwhui, 2021-03-25:
**Describe the bug**

tests/scripts/thread-cert/test_srp_register_single_service.py fails intermittently.

**Console/log output**

- [logs_63743.zip](https://github.com/openthread/openthread/files/6206059/logs_63743.zip)
- [cli-mtd-thread-cert.zip](https://github.com/openthread/openthread/files/6206061/cli-mtd-thread-cert.zip)

wgtdkp, 2021-03-27:
The log shows that the SRP client becomes a `Leader` rather than a `Router`:
```
======================================================================
FAIL: test (__main__.SrpRegisterSingleService)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/examples/../tests/scripts/thread-cert/test_srp_register_single_service.py", line 88, in test
    self.assertEqual(client.get_state(), 'router')
AssertionError: 'leader' != 'router'
```

I can not think of a reason how this could happen. @simonlingoogle Any thoughts?

simonlingoogle, 2021-03-28:
Haven't dig deep into this issue, but I can see some abnormality in the PCAP:

[image]

The SRP client sends two `Parent Requests` quickly, so the second `Parent Request #9` to REED is sent even before the `Parent Response`. This could cause the SRP client failed to process the `Parent Response` because the `challenge` does not match because the SRP client has updated `mParentRequestChallenge` to the challenge  of `Parent Request #9`. 

`Parent Request #9` is sent 200ms after `Parent Request #8`. I am not sure if this is a correct behavior because the timeout is too short (IMO).

jwhui, 2022-03-01:
Have not seen this issue recently. Closing.

closed 2022-03-01 by jwhui