openthread/openthread #6359: PSK questions

purvisb, 2021-03-26:
A colleague of mine is researching the vulnerability of OpenThread to brute force attacks.

What are the character restraints on the PSK?

Also whats the difference between PSKc and PSKd?

jwhui, 2021-03-26:
These details are specified in Chapter 8 of the [Thread 1.1.1 Specification](https://www.threadgroup.org/support#specifications).

In summary:

1. The PSKd (aka Joining Device Credential) is used to establish a secure session between a Commissioner and Joiner. The PSKd is a random value unique to each device. The value is represented as 6 to 32 characters, with an alphabet of 32 characters.
2. The PSKc is the shared secret to establish a secure session between the Commissioner and Thread Border Agent. All nodes in a given Thread network possess the PSKc. This is a 16 byte value.

You can find more details in the Thread Specification.

purvisb, 2021-03-29:
No further questions at this time.
Thanks!

closed 2021-03-29 by purvisb