openthread/openthread #6371: NRF52840 SPI RCP,  RadioSpinelNoResponse

mauricewillemsen, 2021-03-30:
Hello,
We're using openthread daemon on a raspberry pi 3B, and a nordic NRF52840 connected through SPI, on a custom PCB. 

When using ot-daemon compiled with:
`make -f src/posix/bin/Makefile-posix DAEMON=1 RCP_BUS=spi`

And a rcp binary for the nrf chip using:
`make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COAP=1 COMMISSIONER=1 DISABLE_BUILTIN_MBEDTLS=1 DNS_CLIENT=1 DIAGNOSTIC=1 EXTERNAL_HEAP=1 JOINER=1 LINK_RAW=1 MAC_FILTER=1 MTD_NETDIAG=1 SERVICE=1 UDP_FORWARD=1 ECDSA=1 SNTP_CLIENT=1 COAPS=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 NCP_SPI=1`
as suggested by nordic
or 
`make -f examples/Makefile-nrf52840 NCP_SPI=1 `

ot-daemon is started with:
`./ot-daemon -v  'spinel+spi:///dev/spidev0.1?gpio-reset-device=/dev/gpiochip0&gpio-int-device=/dev/gpiochip0&gpio-int-line=21&gpio-reset-line=18&spi-mode=0&spi-speed=8000000&spi-reset-delay=500`

Different SPI settings are tested.

The system works fine for 30min -1hour. And than the following error occurs on ot-daemon:

` [WARN]-PLAT----: radio tx timeout`
`[CRIT]-PLAT----: HandleRcpTimeout() at /home/openthread/src/posix/../../src/lib/spinel/radio_spinel_impl.hpp:2179: RadioSpinelNoResponse`
`[CRIT]-PLAT----: Removing daemon socket: /tmp/openthread.sock`

Radio timeout occurs and ot-daemon stops working. Rerunning always results in the same crash in between 30min - 1hour.

Issue is seen here, and solved with uart-flow-control, however we are using SPI and switching to uart is undesired
https://github.com/openthread/openthread/issues/5662#issuecomment-712731771

Debugging the nrf52840 with a JLink shows that the timer in the log messages is not increased any more after the crash occurs. 
The errors messages that are shown occur all over the time. But the stuck counter happens after the crash.
`[1;36m[0003873568][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003873648][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003875202][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:FcsErr
[1;36m[0003875532][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(72 bytes), error:OK
[1;36m[0003875572][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003877457][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:FcsErr
[1;36m[0003878207][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(63 bytes), error:OK
[1;36m[0003878591][0m [DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
[1;36m[0003878591][0m [DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
[1;36m[0003878602][1;32m [INFO]-MAC-----: Frame tx failed, error:NoAck, retries:0/15, len:113, seqnum:193, type:Data, src:32d
[1;36m[0003878608][0m [DEBG]-MAC-----: RadioState: Transmit -> Receive
[1;36m[0003878608][0m [DEBG]-MAC-----: LinkRaw::TransmitDone(113 bytes), error:OK
[1;36m[0003878738][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879458][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(108 bytes), error:OK
[1;36m[**0003879474**][0m [DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
[1;36m[**0003879474**][0m [DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
[1;36m[**0003879480**][0m [DEBG]-MAC-----: RadioState: Transmit -> Receive
[1;36m[**0003879480**][0m [DEBG]-MAC-----: LinkRaw::TransmitDone(94 bytes), error:OK
[1;36m[**0003879481**][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:Failed
[1;36m[**0003879481**][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[**0003879481**][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(72 bytes), error:OK
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(63 bytes), error:OK
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:NoFrameReceived
[1;36m[0003879481][0m [DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
[1;36m[0003879481][0m [DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(63 bytes), error:OK
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:DestinationAddressFiltered
[1;36m[0003879481][0m [DEBG]-MAC-----: LinkRaw::ReceiveDone(0 bytes), error:Failed`

Anyone has experience with this issue and has a solution?

abtink, 2021-04-07:
I can suggest some ideas to may be try out and investigate.

- What is the NCP buffer size on the RCP side? (`OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE`)?
   - If you can change/increase/decrease it, it may be be good to see if impacts the issue (happen less frequently, or not)?

- Do you have logs enabled on RCP side? 
  - If so, can you disable logging on RCP to see if it helps or not?

- When we see the error on host with `HandleRcpTimeout()` do we see any other events from RCP?
   - E.g., do we see an RCP reset event? or is it always no response timeout?

mauricewillemsen, 2021-04-07:
Thanks for your response,

- Did increase the `OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE`, crash still occurs. Will investigate more to see the impact.
- Tried both with logs enabled and disabled. Didnt solve the problem
- The error on the host is always the tx timeout which results in no response timeout.

mauricewillemsen, 2021-04-07:
Crash occurs faster with both decreasing and increasing `OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE`.

jwhui, 2021-04-13:
@mauricewillemsen , are you able to reproduce this issue with a Nordic dev board or USB dongle?

mauricewillemsen, 2021-04-21:
@jwhui , The issue seems related to SPI, cause it is not happening with the USB dongles.
However we have custom hardware board, with only SPI exposed to raspberry pi, and not uart. Having USB dongles is undesirable as well.
Setting `RESTORATION_MAX_COUNT` gives that the RCP is restored and the ot-daemon is not killed. 
But the issue that the restoration is necessary remains.

We've found that is proportional to the traffic count. When performing fw-update through the network with quite some devices. The RCP needs to restore every minute. In normal circumstances +/- 30min

pdgendt, 2021-04-21:
I can confirm this issue, I've had the same problem with an nRF52840 RCP SPI connected to posix border router (custom board).

However I also had UART available, which never seemed to fail, so we changed to that.

jwhui, 2021-04-21:
@LuDuda @MarekPorwisz , is this something you can help take a look at?

MarekPorwisz, 2021-04-22:
@pszkotak Please have a look at it

pszkotak, 2021-04-27:
@mauricewillemsen, so far I did not not manage to reproduce the issue with the ot-cli (instead of ot-daemon) with lower clock (500kHz instead of 8MHz). Now I'm switching to the values you tested. It would make debugging a bit harder, but there I do not see other options at this moment.

In my test scenario I use another CLI FTD device which is constantly pinging the RCP+RPi node.
The command I use is `ping <ip> 100 65000`. Please let me know if this is stressful enough and reproduces the issue in your testing as well. My goal is to keep the test environment as simple as possible.

In the meantime I'm switching to the ot-daemon + 8MHz clock.

pszkotak, 2021-04-27:
It's been at least 3 hours with the ot-deamon and 8MHz SPI clock and still no issues with the communication.
@mauricewillemsen, do you have any more hints that could help to reproduce the issue?

pszkotak, 2021-04-28:
Another question that comes to my mind is about the connection.
Have you connected the interrupt pin? As far as I remember we experienced some issues without it in the past.
In my case the connection looks exactly as [here](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/thread_border_router.html#thread_border_router_setup_3_spi).

pszkotak, 2021-04-30:
@MarekPorwisz, unassigning myself. Please reassign if there will be a need for it.

Endag, 2021-05-05:
> @jwhui , The issue seems related to SPI, cause it is not happening with the USB dongles.
 I've ran into same issue with RCP connected by USB, so it seems to be something more to  investigate

jwhui, 2021-05-05:
> I've ran into same issue with RCP connected by USB, so it seems to be something more to investigate
> 
> More to say it stays stable until you try to get status ot any other information via ot-ctl, then it enters restart-timeout cycle

@Endag , this seems like a different issue. If you continue to experience this problem, please open a new GitHub Issue and provide more details necessary to reproduce the issue, including git commit, commands used to build the firmware image, etc.

Endag, 2021-05-06:
> @Endag , this seems like a different issue. If you continue to experience this problem, please open a new GitHub Issue and provide more details necessary to reproduce the issue, including git [...]

gotcha, moved with my problem https://github.com/openthread/openthread/issues/6575

jwhui, 2021-05-06:
@mauricewillemsen , any thoughts on the questions posed above?

- https://github.com/openthread/openthread/issues/6371#issuecomment-827822428
- https://github.com/openthread/openthread/issues/6371#issuecomment-828226297

jwhui, 2021-05-27:
Closing stale issue.

closed 2021-05-27 by jwhui