openthread/openthread #6391: Custom TLVs in network diagnostics response

dismirlian, 2021-04-01:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
Thread's network diagnostics feature is very useful to get information from devices, from the Thread network itself, topology visualization, etc., to help manage the deployment. However, the device's application may also need to report relevant metrics. These can be related to the network (eg. receiver average rssi), to the device itself (eg. eui64), to the device's "operating system" (eg. free memory), or even application specific.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
The application can expose a CoAP resource with the required extra diagnostics information.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
Allow custom TLVs in the network diagnostics response (or maybe a single "Custom TLV" with application-specific sub-TLVs).

**Additional context** Add any other context or screenshots about the feature request here.
Although exposing a separate resource with the required extra diagnostics information is easier, I think in large networks it's useful to avoid the extra message, especially if using multicast DIAG_GET.qry.

Implementation-wise, I think this can be done with a callback mechanism in `FillRequestedTlvs`, here:
https://github.com/openthread/openthread/blob/7dca56e9821dcf6ee0825a67a056d44ddf47075d/src/core/thread/network_diagnostic.cpp#L444-L447

What do you think?

dismirlian, 2021-04-01:
As a side-effect of this proposal, two TLVs covered by the spec can be filled by the callback: battery level TLV and supply voltage TLV.

Irving-cl, 2021-04-08:
@dismirlian 

> Implementation-wise, I think this can be done with a callback mechanism in FillRequestedTlvs,

Do you mean to provide an API in OT for the application to set a callback to get the custom values for custom TLV? I think the network diagnositcs interface should be unified. For example, if I have 2 products (A and B) from 2 different vendors who defined different custom TLVs, when A makes query B, it gets a custom TLV (but implemented by B). So the data in the custom TLV doesn't make any sense to A. 

I think if the application wants extra diagnositics information, probably it should collect the information itself.

dismirlian, 2021-04-12:
Hi @Irving-cl, thanks for your answer.

> Do you mean to provide an API in OT for the application to set a callback to get the custom values for custom TLV?

Yes. I agree it should be unified, but it might help in case of a "closed" deployment.

Our particular use case was dealt with using an application-specific CoAP endpoint, so I'll close this issue for now...

Thanks for your support.

closed 2021-04-12 by dismirlian