openthread/openthread #6405: Please synchronize the image tool of K32W061 with SDK_2.6.2_K32W061DK6_MR2_1

JumaLuata, 2021-04-06:
The image tool of SDK_2.6.2_K32W061DK6_MR2_1 has been upgraded to support python3. 
On Ubuntu 20.04, the python-pip deb package cannot be installed, which will cause errors when verifying `is_linux_package_installed "python-pip"` when running the **_sign_image.sh_** script.
You can get SDK_2.6.2_K32W061DK6_MR2_1 from [here](https://mcuxpresso.nxp.com/en/select).

doru91, 2021-04-06:
Hi, @JumaLuata 

Thanks for raising this issue, we are aware of it. It will be solved shortly: we are in progress of refactoring the OpenThread Code and in the future, we'll use directly the SDK files (through an env variable) instead of using the old ones from third_party/nxp.

I'll post an update by the end of the week with the new refactored repository.

If you are an internal NXP employee please contact me internally.

Br,
Doru

jwhui, 2021-04-09:
NXP examples have moved to https://github.com/openthread/ot-nxp

closed 2021-04-09 by jwhui