openthread/openthread #6411: Developer Meeting: TCP in OpenThread

jwhui, 2021-04-06:
The OpenThread project will be hosting its first developer meeting the week of April 12th, 2021.

**Topic:** Transport Control Protocol (TCP) in OpenThread

**Proposed Agenda**
1. TCP APIs
2. TCP Implementation

**Doodle Poll**

Anyone interested in discussing TCP in OpenThread is welcome to participate. If you would like to participate, please respond to this Doodle poll:
- https://doodle.com/poll/2zty8hnrea7xnhhk?utm_source=poll&utm_medium=link

Please subscribe to this GitHub issue for further updates on the developer meeting.

jwhui, 2021-04-09:
The Doodle poll is now closed. The most voted for time slot was Thursday April 15th 08:00 PDT, 11:00 EDT, 15:00 UTC, 17:00 CEST, 23:00 CST.

If you would like to participate in this meeting, please send me your name and email address and I will add you to the calendar invite.

jwhui, 2021-04-13:
This is a reminder that this meeting will be held Thursday April 15th 08:00 PDT, 11:00 EDT, 15:00 UTC, 17:00 CEST, 23:00 CST.

If you would like to participate in this meeting, please send me your name and email address and I will add you to the calendar invite.

samkumar, 2021-04-15:
Here are the slides that I presented at today's meeting: [openthread_tcp_design_api.pdf](https://github.com/openthread/openthread/files/6319431/openthread_tcp_design_api.pdf)

At the next meeting, I'll present some slides discussing possibilities for TCP implementations, and we'll have a discussion about TCP implementations similar to the one we had today about TCP APIs.

beriberikix, 2021-04-15:
Apologies for missing the presentation and if this was brought up in
person, but the slides mention CoAP doesn't support high-bandwidth
transfers but that isn't entirely accurate. To clarify, RFC 8323
<https://tools.ietf.org/html/rfc8323> defines TCP as a transport and is
implemented in projects like libcoap <https://github.com/obgm/libcoap>.
Additionally, the CoRE working group specified block transfers (RFC 7959
<https://tools.ietf.org/html/rfc7959>) and newer block types for larger
amounts of data (draft-ietf-core-new-block
<https://core-wg.github.io/new-block/draft-ietf-core-new-block.html>.)

That being said, support for TCP would be a welcome addition and the
presentation makes a strong case!

Cheers.

samkumar, 2021-04-15:
Thanks @beriberikix for raising this point. I should clarify that I was referring to UDP-based CoAP since that's the only way that CoAP can be used as an alternative to TCP. I'm aware of the block-transfer protocol in CoAP, but my understanding is that each block must be ACKed before the next block can be sent, meaning that throughput degrades inversely with the network round-trip time. TCP throughput does not degrade in the same way because it allows multiple in-flight segments for a single connection. That's what I was referring to when I said that CoAP has lower throughput.

Thanks for sharing the information about the new "Quick Blocks" feature. I wasn't previously aware of that. I agree that it would allow high-bandwidth CoAP. It uses similar mechanisms as TCP (e.g., multiple packets in flight simultaneously), and I think that it could co-exist with TCP as an option in these networks.

closed 2021-05-01 by jwhui