openthread/openthread #6437: Is OpenThread compatible with nrf51822?

purvisb, 2021-04-12:
I'm exploring the idea of getting OpenThread working on a [Crazyflie drone](https://store.bitcraze.io/products/crazyflie-2-0).

Is OpenThread compatible with the nrf51822 platform? I don't see a MakeFile for it in openthread/examples/.  I'm not sure if this is an indication of incompatibility. Or if there just isn't a make file for that.

jwhui, 2021-04-12:
@purvisb , Thread currently requires an IEEE 802.15.4 radio. I believe the nRF51822 does not support IEEE 802.15.4.

purvisb, 2021-04-12:
@jwhui The [nRF51822](https://www.nordicsemi.com/Products/Low-power-short-range-wireless/nRF51822) has BLE. Doesn't BLE run on 802.15.4?

jwhui, 2021-04-12:
@purvisb , BLE is built on 802.15.1.

purvisb, 2021-04-12:
@jwhui thank you for the clarification!

closed 2021-04-12 by purvisb