openthread/openthread #6456: Developer Meeting: TCP Implementation

jwhui, 2021-04-15:
The OpenThread project will be hosting its next developer meeting on Thursday April 29th.

**Topic:** Transport Control Protocol (TCP) in OpenThread

**Time:** Thursday April 29th 08:00 PDT, 11:00 EDT, 15:00 UTC, 17:00 CEST, 23:00 CST.

**Agenda**

TCP Implementation

**Meeting Info**

If you would like to participate in this meeting, please send me your name and email address and I will add you to the calendar invite.

backkem, 2021-04-27:
Hi @jwhui, I'm wondering if this meeting is open to the public. I'm trying to broaden LAN support in browsers, see [Local devices API](https://discourse.wicg.io/t/idea-local-devices-api-lan-services). Enabling secure access to Thread devices from the browser would be a perfect use-case.

jwhui, 2021-04-27:
> Hi @jwhui, I'm wondering if this meeting is open to the public. I'm trying to broaden LAN support in browsers, see [Local devices API](https://discourse.wicg.io/t/idea-local-devices-api-lan-services). Enabling secure access to Thread devices from the browser would be a perfect use-case.

@backkem , yes, this meeting is open to anyone in the OpenThread community that would like to join. Please send me your name and email address and I will add you to the calendar invite.

backkem, 2021-04-27:
@jwhui Michiel De Backker; [image]

samkumar, 2021-04-29:
Here are the slides that I presented at today's meeting: [openthread_tcp_design_implementation.pdf](https://github.com/openthread/openthread/files/6400253/openthread_tcp_design_implementation.pdf)

There was broad support for using TCPlp. As we discussed, I'll make a pull request with the entire TCP stack so that people can try out the code, but I'll also submit "bite-sized" pull requests so that the code can be reviewed and merged incrementally.

romacdon, 2021-05-06:
> Here are the slides that I presented at today's meeting: [openthread_tcp_design_implementation.pdf](https://github.com/openthread/openthread/files/6400253/openthread_tcp_design_implementation.pdf) [...]

Hi @samkumar - thanks for all the research and the great presentations wrt to your TCP work. This IETF document was recently brought to my attention and I'm wondering how much the info in this doc overlaps with the research you've done. https://tools.ietf.org/id/draft-ietf-lwig-tcp-constrained-node-networks-04.html.

samkumar, 2021-05-06:
Hi @ramacdon, that Internet Draft is indeed in the same spirit as my research. The same author, Prof. Gomez, also published an [article in IEEE Internet Computing](https://ieeexplore.ieee.org/document/8259430) that you may be interested in.

Overall, these are the key differences I've identified:
1. The Internet Draft gives recommendations on TCP usage, and additional useful TCP features, in the context of simplified TCP stacks that don't support all TCP features. For example, many of the recommendations are about so-called "Single MSS" implementations that can only have one packet in flight. In contrast, I use a full-scale TCP implementation (TCPlp), which makes this a non-issue.
    * An important observation we made in our research is that full-scale TCP's memory footprint is reasonable for a wide range of devices. That said, I was focusing on a similar class of devices that Thread/OpenThread targets --- "Class 2" devices according to [RFC 7228, Section 3](https://tools.ietf.org/html/rfc7228#page-8) --- while the Internet Draft may be considering a broader set of devices. If you're using devices that are severely resource constrained (e.g., 10 KiB of RAM or less), too resource-constrained to run OpenThread, then the approach in the Internet Draft might be a better option than using a full-scale TCP stack.
2. My research is based on a set of comprehensive experiments in a real IEEE 802.15.4 testbed and a set of performance goals and measurements, whereas the Internet Draft (to my knowledge) is based on reasoning about how each TCP feature behaves. This means that, in my research:
    * We can make specific recommendations about parameter values, like what MSS values works well.
    * We identify and discuss issues regarding interaction with the link-layer, such as the hidden terminal problem degrading throughput and link-layer scheduling issues in Thread, and TCP's congestion control behaving differently. These issues aren't discussed in the Internet draft, perhaps because they aren't obvious based on reasoning, but are impossible to ignore when you do an actual experiment.
    * We can validate that the resulting TCP implementation works well when our recommendations are used.

romacdon, 2021-05-06:
Thank you very much for the detailed reply @samkumar.

closed 2021-07-12 by jwhui