openthread/openthread #6460: [codecov] add upload commands to GitHub Action Workflow files

jwhui, 2021-04-16:
Upload was removed from `./script/test` in #6459 due to security concerns.

See https://about.codecov.io/security-update/

jinran-google, 2021-04-16:
The Codecov GitHub Action is using the same bash script, without any integrity checks, so it should not be an option: <https://github.com/codecov/codecov-action/blob/9b0b9bbe2c64e9ed41413180dd7398450dfeee14/src/index.ts#L15>

I think either we have to keep coverage uploading disabled for a while, until a good solution occurs, or we save a copy of the bash script in our repo and manually update it in the future.

jwhui, 2021-04-16:
> The Codecov GitHub Action is using the same bash script, without any integrity checks, so it should not be an option: [...]

@jinran-google , the primary concern is using it in scripts that a developer may run locally. Using the Codecov GitHub Action does not have this same risk.

jinran-google, 2021-04-16:
> @jinran-google , the primary concern is using it in scripts that a developer may run locally. Using the Codecov GitHub Action does not have this same risk.

OK, that makes sense. It's OK to use their GitHub Action as we don't use any other secrets besides `secrets.GITHUB_TOKEN` in GitHub Actions.

Also, Codecov is implementing a checksum verification in their GitHub Action and it will be landed by the end of tomorrow.

closed 2021-04-19 by jwhui