openthread/openthread #6462: Failed to build POSIX `ot-ctl` with TREL link only (without 802.15.4)

simonlingoogle, 2021-04-16:
**Describe the bug**

Failed to build on POSIX `ot-ctl` with TREL link only.

**To Reproduce** 
```bash
$ script/cmake-build posix -DOT_TREL=ON -DCMAKE_CXX_FLAGS="-DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0"
```

**Expected behavior**
Build ot-ctl successfully.

**Console/log output** 
```
[11/483] Building CXX object src/posix/CMakeFiles/ot-ncp.dir/ncp.cpp.o
FAILED: src/posix/CMakeFiles/ot-ncp.dir/ncp.cpp.o 
/usr/bin/c++ -DMBEDTLS_CONFIG_FILE=\"openthread-mbedtls-config.h\" -DOPENTHREAD_POSIX_APP_TYPE=OT_POSIX_APP_TYPE_NCP -DOPENTHREAD_POSIX_CONFIG_MAX_POWER_TABLE_ENABLE=1 -Ietc/cmake -I../../etc/cmake -I../../include -I../../src -I../../src/core -I../../src/posix/platform -I../../src/posix/platform/include -I../../third_party/mbedtls/repo/include -DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0 -g -Werror -pedantic-errors -Wall -Wextra -Wshadow -Wno-c++14-compat -fno-exceptions -std=gnu++11 -MD -MT src/posix/CMakeFiles/ot-ncp.dir/ncp.cpp.o -MF src/posix/CMakeFiles/ot-ncp.dir/ncp.cpp.o.d -o src/posix/CMakeFiles/ot-ncp.dir/ncp.cpp.o -c ../../src/posix/ncp.cpp
In file included from ../../src/core/openthread-core-config.h:80,
                 from ../../src/posix/platform/openthread-posix-config.h:32,
                 from ../../src/posix/ncp.cpp:29:
../../src/core/config/radio_link.h:61:2: error: #error "No radio link type is enabled - at least one radio link type should be enabled"
   61 | #error "No radio link type is enabled - at least one radio link type should be enabled"
      |  ^~~~~
[14/483] Building CXX object src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o
FAILED: src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o 
/usr/bin/c++ -DHAVE_LIBREADLINE=1 -DMBEDTLS_CONFIG_FILE=\"openthread-mbedtls-config.h\" -DOPENTHREAD_POSIX_APP_TYPE=OT_POSIX_APP_TYPE_CLI -DOPENTHREAD_POSIX_CONFIG_MAX_POWER_TABLE_ENABLE=1 -DOPENTHREAD_SPINEL_CONFIG_OPENTHREAD_MESSAGE_ENABLE=0 -Ietc/cmake -I../../etc/cmake -I../../include -I../../src -I../../src/core -I../../src/posix/platform -I../../src/posix/platform/include -I../../third_party/mbedtls/repo/include -DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0 -g -Werror -pedantic-errors -Wall -Wextra -Wshadow -Wno-c++14-compat -fno-exceptions -std=gnu++11 -MD -MT src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o -MF src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o.d -o src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o -c ../../src/posix/cli.cpp
In file included from ../../src/core/openthread-core-config.h:80,
                 from ../../src/posix/platform/openthread-posix-config.h:32,
                 from ../../src/posix/cli.cpp:29:
../../src/core/config/radio_link.h:61:2: error: #error "No radio link type is enabled - at least one radio link type should be enabled"
   61 | #error "No radio link type is enabled - at least one radio link type should be enabled"
      |  ^~~~~
```

The following command can work:
```
$ script/cmake-build posix -DOT_TREL=ON -DCMAKE_CXX_FLAGS="-DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0 -DOPENTHREAD_CONFIG_RADIO_LINK_TREL_ENABLE=1"
```

jwhui, 2021-04-16:
Yes, multi radio requires at least one of:
- `-DOPENTHREAD_CONFIG_RADIO_LINK_TREL_ENABLE=1`
- `-DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=1`

simonlingoogle, 2021-04-16:
@jwhui 
Yes. But see I have already used `-DOT_TREL=ON`. Seems it does not pass configuration `OPENTHREAD_CONFIG_RADIO_LINK_TREL_ENABLE=1` to posix apps.

abtink, 2021-04-16:
When building the posix platform configs seems to be set up correctly.
See we have both `-DOPENTHREAD_CONFIG_RADIO_LINK_TREL_ENABLE=1` and `-DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0`
```
[144/568] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ 
[-DOPENTHREAD_CONFIG_ASSERT_ENABLE=1 -DOPENTHREAD_CONFIG_BACKBONE_ROUTER_DUA_NDPROXYING_ENABLE=0
[-DOPENTHREAD_CONFIG_BACKBONE_ROUTER_MULTICAST_ROUTING_ENABLE=0 -DOPENTHREAD_CONFIG_BORDER_AGENT_ENABLE=1
[-DOPENTHREAD_CONFIG_BORDER_ROUTER_ENABLE=1 -DOPENTHREAD_CONFIG_CHANNEL_MANAGER_ENABLE=1
[-DOPENTHREAD_CONFIG_CHANNEL_MONITOR_ENABLE=1 -DOPENTHREAD_CONFIG_CHILD_SUPERVISION_ENABLE=1
[-DOPENTHREAD_CONFIG_COAP_API_ENABLE=1 -DOPENTHREAD_CONFIG_COAP_BLOCKWISE_TRANSFER_ENABLE=1
[-DOPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE=1 -DOPENTHREAD_CONFIG_COMMISSIONER_ENABLE=1
[-DOPENTHREAD_CONFIG_DATASET_UPDATER_ENABLE=1 -DOPENTHREAD_CONFIG_DHCP6_CLIENT_ENABLE=1
[-DOPENTHREAD_CONFIG_DHCP6_SERVER_ENABLE=1 -DOPENTHREAD_CONFIG_DIAG_ENABLE=1 -DOPENTHREAD_CONFIG_DNS_CLIENT_ENABLE=1
[-DOPENTHREAD_CONFIG_ECDSA_ENABLE=1 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=1
[-DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=1 -DOPENTHREAD_CONFIG_FILE=\"openthread-core-posix-config.h\"
[-DOPENTHREAD_CONFIG_IP6_FRAGMENTATION_ENABLE=1 -DOPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1
[-DOPENTHREAD_CONFIG_JAM_DETECTION_ENABLE=1 -DOPENTHREAD_CONFIG_JOINER_ENABLE=1 -DOPENTHREAD_CONFIG_LEGACY_ENABLE=1
[-DOPENTHREAD_CONFIG_LOG_LEVEL_DYNAMIC_ENABLE=1
[-DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -DOPENTHREAD_CONFIG_MAC_FILTER_ENABLE=1
[-DOPENTHREAD_CONFIG_NCP_HDLC_ENABLE=1 -DOPENTHREAD_CONFIG_NEIGHBOR_DISCOVERY_AGENT_ENABLE=1
[-DOPENTHREAD_CONFIG_PING_SENDER_ENABLE=1 -DOPENTHREAD_CONFIG_RADIO_LINK_TREL_ENABLE=1
[-DOPENTHREAD_CONFIG_REFERENCE_DEVICE_ENABLE=1 -DOPENTHREAD_CONFIG_SNTP_CLIENT_ENABLE=1
[-DOPENTHREAD_CONFIG_SRP_CLIENT_ENABLE=1 -DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_2
[-DOPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_ENABLE=1 -DOPENTHREAD_CONFIG_TMF_NETWORK_DIAG_MTD_ENABLE=1
[-DOPENTHREAD_ENABLE_COVERAGE=1 -DOPENTHREAD_FTD=1 -DOPENTHREAD_MTD=0 -DOPENTHREAD_POSIX_CONFIG_MAX_POWER_TABLE_ENABLE=1
[-DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-posix-config.h\"
[-DOPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT=2 -DPACKAGE_NAME=\"OPENTHREAD\"
[-DPACKAGE_VERSION=\"20191113-01921-gf81c0b50f-dirty\" -Ietc/cmake -I../../etc/cmake -I../../include -I../../src
[-I../../src/core -I../../third_party/mbedtls/repo/include -I../../src/posix/platform/include -I../../src/posix/platform
[-Ithird_party/mbedtls -DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0 -g -isysroot
[/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.0.sdk   -Werror
[-pedantic-errors -Wall -Wextra -Wshadow -Wno-c++14-compat -fno-exceptions -g -O0 --coverage -std=gnu++11 -MD -MT
[src/posix/platform/CMakeFiles/openthread-posix.dir/trel_udp6.cpp.o -MF
[src/posix/platform/CMakeFiles/openthread-posix.dir/trel_udp6.cpp.o.d -o
[src/posix/platform/CMakeFiles/openthread-posix.dir/trel_udp6.cpp.o -c ../../src/posix/platform/trel_udp6.cpp```

abtink, 2021-04-16:
I spent some time debugging this (enabling `verbose` mode on cmake and `ninja`). 

I think this has to do with how cmake files were declared for posix app executables. _It actually has nothing to do with TREL and the `OT_TREL` flag itself._

See the command here for building `../../src/posix/cli.cpp`. None of the configs are incldued here other than the `DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE` (which is manually added). 
```
[145/568] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
[ -DHAVE_LIBREADLINE=1 -DMBEDTLS_CONFIG_FILE=\"openthread-mbedtls-config.h\"
[-DOPENTHREAD_POSIX_CONFIG_MAX_POWER_TABLE_ENABLE=1
[-DOPENTHREAD_SPINEL_CONFIG_OPENTHREAD_MESSAGE_ENABLE=0 -Ietc/cmake -I../../etc/cmake
[-I../../include -I../../src -I../../src/core -I../../src/posix/platform
[-I../../src/posix/platform/include -I../../third_party/mbedtls/repo/include
[-DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0 -g -isysroot
[/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.0.sdk 
[ -Werror -pedantic-errors -Wall -Wextra -Wshadow -Wno-c++14-compat -fno-exceptions -std=gnu++11 -MD
[-MT src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o -MF src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o.d -o
[src/posix/CMakeFiles/ot-cli.dir/cli.cpp.o -c ../../src/posix/cli.cpp
```

abtink, 2021-04-17:
As I mentioned the issue is more fundamental to how the `posix` cmake files are defined and it is not really tied to TREL or `OT_TREL` flag.

The issue seems to be with  `src/posix/cli.cmake` (same situation with `src/posix/daemon.cmake`). They don't include `ot-config` under `target_link_libraries` and therefore the `OEPNTHREAD_CONFIG` definitions from cmake options are not included when building files under that target (`cli.cpp`, `main.cpp`, etc).

I submitted PR https://github.com/openthread/openthread/pull/6474 as a potential fix for this. With this change, this command works fine:
```
script/cmake-build posix -DOT_TREL=ON -DCMAKE_CXX_FLAGS="-DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0"
```

But I am noticing with the change in this PR, one specific test keeps failing after multiple-re-runs: `POSIX` workflow test `pty-macos OT_DAEMON=on` ([link to a run](https://github.com/openthread/openthread/pull/6474/checks?check_run_id=2366988420)).

I don't want to spend more time on this. So leave this for the folks who know more on `posix` cmake file and what the the related failing test does to investigate the reason for failure.

@simonlingoogle , you can use the command below (which as you point out works). 
```
script/cmake-build posix -DCMAKE_CXX_FLAGS="-DOPENTHREAD_CONFIG_RADIO_LINK_IEEE_802_15_4_ENABLE=0 -DOPENTHREAD_CONFIG_RADIO_LINK_TREL_ENABLE=1"
```

simonlingoogle, 2021-04-19:
Any comment on this issue? @bukepo

jwhui, 2021-04-28:
Resolved by #6474

closed 2021-04-28 by jwhui