openthread/openthread #6479: linkmetrics not working as I expect

purvisb, 2021-04-19:
**Describe the bug** 
Here's my outputs
```
> linkmetrics query fe80:0:0:0:1029:e885:cfe7:de20 single qmr
done
```
```
> linkmetrics mgmt fe80:0:0:0:1029:e885:cfe7:de20 enhanced-ack register qmr
Received Link Metrics Management Response from: fe80:0:0:0:1029:e885:cfe7:de20
Status: Other error
```
**To Reproduce**

Git commit id: de93711e223aade9867d4bf196258776be269b79
Using nrf52840-mdk
build all firmware with `USB=1 THREAD_VERSION=1.2 LINK_METRICS=1 JOINER=1 COMMISSIONER=1`

**Expected behavior** 
The same outputs as seen [here](https://github.com/openthread/openthread/blob/main/src/cli/README.md#linkquality-extaddr)

**Console/log output**
[linkmetric pcaps.zip](https://github.com/openthread/openthread/files/6332778/linkmetric.pcaps.zip)

Irving-cl, 2021-04-19:
Hi @purvisb 

Thanks for raising the issue. There are a few problems here:

* Your commit `de93711` is a little old. There are some fixes to this issue after this commit. Please use the latest commit.
* Regarding `linkmetrics mgmt fe80:0:0:0:1029:e885:cfe7:de20 enhanced-ack register qmr
`, please note that according to spec, **at most two** metrics can be used in enhanced-ack probing. So you can do `linkmetrics mgmt fe80:0:0:0:1029:e885:cfe7:de20 enhanced-ack register qm` or `linkmetrics mgmt fe80:0:0:0:1029:e885:cfe7:de20 enhanced-ack register mr`, etc. These should work. We also mentioned it in our cli doc (Probably we should bold it):

[image]
* If you do the query from a child to a parent, it should work (with the 2 points I mentioned above). If you do the query from a parent to a child, there is an issue. I fixed the issue in #6483.

Please have a try and let me know :)

jwhui, 2021-05-01:
Closing stale issue.

closed 2021-05-01 by jwhui