openthread/openthread #6490: Duplicate echo replies

superwhd, 2021-04-20:
**Describe the bug** 

A router sends an echo request to a border router. Then the border router sends back two echo replies.

**To Reproduce**

I detected this error in simulation environment. See https://github.com/openthread/openthread/pull/6484.

```
$ ./script/test build_otbr_docker && THREAD_VERSION=1.2 VIRTUAL_TIME=0 REFERENCE_DEVICE=1  ./script/test build
$ sudo PACKET_VERIFICATION=1 VERBOSE=1 THREAD_VERSION=1.2 VIRTUAL_TIME=0 ./script/test cert_suite tests/scripts/thread-cert/test_ping.py
```

**Expected behavior** 

This is a unicast echo request so I expected only one reply will be sent for every request.

**Console/log output** 

**Additional context**

simonlingoogle, 2021-04-20:
@superwhd 
Thanks for revealing and reproducing this issue. 

I guess maybe the ping request is handled by both OpenThread and Linux Kernel, causing duplicate ping responses.

jwhui, 2022-09-13:
Closing stale issue.

closed 2022-09-13 by jwhui