openthread/openthread #6494: IP packet from link-local address is forwarded out of the link

superwhd, 2021-04-20:
**Describe the bug**

Topology: ROUTER_2 ----- ROUTER_1 ---- ROUTER_3

ROUTER_2 sends an echo request, from ROUTER_2's link-local address to ROUTER_3's RLOC.

The ping request is forwarded to ROUTER_3, which I think is not as intended.

**To Reproduce** 

The test case is in https://github.com/openthread/openthread/pull/6492.

To run the test case:
```
$ ./script/test build
$ PACKET_VERIFICATION=1 VERBOSE=1  ./script/test cert_suite tests/scripts/thread-cert/test_ping.py 
```

**Expected behavior**
The ping packet shouldn't be forwarded to ROUTER_3.

**Console/log output**
[image]

**Additional context**

jwhui, 2021-04-21:
Thread routers forward IPv6 messages across multiple hops at the 6LoWPAN layer - a layer below IP. In this case, I do not think it is the job of ROUTER_1 to drop the message. ROUTER_1 is simply forwarding the 6LoWPAN frame as is specified by the Mesh Addressing header.

We could add some code in the originating router (ROUTER_2 in this case) to drop the packet if it notices that the destination is more than one hop away. However, I'm not sure if there's a real benefit to doing so.

Thoughts?

simonlingoogle, 2021-04-22:
> Thread routers forward IPv6 messages across multiple hops at the 6LoWPAN layer - a layer below IP. In this case, I do not think it is the job of ROUTER_1 to drop the message. ROUTER_1 is simply forwarding the 6LoWPAN frame as is specified by the Mesh Addressing header.

I think you are right that the Router is forwarding the message according to its mesh header.

> We could add some code in the originating router (ROUTER_2 in this case) to drop the packet if it notices that the destination is more than one hop away. However, I'm not sure if there's a real benefit to doing so.

Maybe we just say it's expected behavior as long as the ping is not successful in the end.

@librasungirl @superwhd

simonlingoogle, 2021-04-25:
Agree with @jwhui that checking the source address is not that useful:

- The checking code is never useful but only adds code size because OT should never sends packets with wrong source addresses (except for testing purpose)
- In the case of testing, we actually prefer the packets with wrong source addresses to be sent out so as to test the behavior of the receiving devices.

Should we close this issue and #6492 since it's expected behavior?

@librasungirl @superwhd

librasungirl, 2021-05-11:
sgtm to keep it as is now.

closed 2021-05-11 by librasungirl