openthread/openthread #6526: The link is invalid

Tonyyengithub, 2021-04-28:
**Describe the bug** A clear and concise description of what the bug is.
Page (https://openthread.io/codelabs/openthread-hardware#1)
'Install nRF5x Command Line Tools' website link is invalid

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

Modify the provided link

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.
I think the right link might be 'https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Command-Line-Tools/Download'
This is my first time using github to submit the bug, if there's any offense or incorrect please tell me, many thanks.

Vyrastas, 2021-04-29:
Thanks for reporting this, I'll check across all the codelabs and docs to ensure the link is updated, as I think we use it in other content.

Vyrastas, 2021-05-04:
This will be fixed in https://github.com/openthread/ot-docs/pull/14

jwhui, 2021-05-27:
@Tonyyengithub , this has been fixed. Thanks again for reporting this issue!

closed 2021-05-27 by jwhui