openthread/openthread #6528: Node not acquiring address from on-mesh(global) prefix.

Endag, 2021-04-28:
In my test network i have border gateway based on otbr( commit 8f23e77966a36ed67eb99b9be14e6189806977b6) and two nodes, one based on stock nRF52840 ftd-cli (from 4.1.1 SDK), second is my code loosely based on that SDK.

Demo from SDK acquiring global address normally, but my own application is not.

What kind of settings can cause such behavior? I tried to not modify openthread's config files from SDK when importing to my project. (Enabled CoAP and some platform changes)

Cannot provide steps to reproduce because my code is my employer property.

wgtdkp, 2021-04-28:
I suspect your second end device is not in the same Thread network which is
formed by OTBR:
1. make sure that you can ping the OTBR device from your second end device.
2. please provide the result of command "netdata show" on all the 3 devices
if pinging OTBR is working.

BRs,
kangping

Endag, 2021-04-28:
I can all nodes by mesh-local address normally in any combination.

Openthread release in my nodes is too old to give human-readable network data, will binary dump be helpful to you?

Endag, 2021-04-28:
I'll leave it here anyway to save time

On first glance end-devices netdata is matching

ot-ctl netdata show
Prefixes:
fd11:22:0:0::/64 paros med d800
Routes:
fd5d:d8ff:4288:0::/64 s med d800
Services:
Done

ftd-cli
03140040fd110022000000000504d800330007021140030f0040fd5dd8ff428800000103d80000081a0b02d4010810000000000000000000000000000000000902d800

my app
03140040fd110022000000000504d800330007021140030f0040fd5dd8ff428800000103d80000081a0b02d4010810000000000000000000000000000000000902d800

jwhui, 2021-04-28:
@Endag , one thing to check is whether [OPENTHREAD_CONFIG_IP6_SLAAC_ENABLE](https://github.com/openthread/openthread/blob/2999f834e644cd84ad5991a104f736cf1da39ce0/src/core/config/ip6.h#L117) is enabled.

Endag, 2021-04-29:
Enabling SLAAC indeed helps, thank you.

But I didn't find in documentation any clear clues that SLAAC is needed for communication beyond scope of mesh, as  mesh-local address is auto-assigned by stack when commissioning i was expecting that same goes for global address, if prefix available.
Perhaps documentation (or at least config entries in headers) need some clarification.

closed 2021-04-29 by Endag