openthread/openthread #6538: [github-actions] test_ndproxy.py fails initermittently

jwhui, 2021-04-29:
[logs_72922.zip](https://github.com/openthread/openthread/files/6402091/logs_72922.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6402092/thread-1-2-backbone-results.zip)

jwhui, 2021-05-05:
[logs_8082.zip](https://github.com/openthread/openthread/files/6430526/logs_8082.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6430527/thread-1-2-backbone-results.zip)

simonlingoogle, 2021-05-07:
**Current findings:***

PBBR (also Leader) re-register its DUA and sends unsolicited NA and `Backbone Answer`, which resets its `Time Since Last Transaction` to zero.

```
Apr 29 22:49:00 05d19bedf796 otbr-agent[104]: [INFO]-DUA-----: Sent DUA.req for DUA fd00:7d03:7d03:7d03:15e7:e396:34d6:4d81: OK
Apr 29 22:49:00 05d19bedf796 otbr-agent[104]: [INFO]-NDPROXY-: NdProxyManager: SendNeighborAdvertisement: OK
Apr 29 22:49:00 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: Send PRO_BB.ntf for fd00:7d03:7d03:7d03:15e7:e396:34d6:4d81 (rloc16=fffe): OK
Apr 29 22:49:00 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: NdProxyTable::Register 15e7e39634d64d81 MLIID f08948bc3848c95b RLOC16 f400 LTT 0 => OK
Apr 29 22:49:00 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: Received DUA.req on PBBR: OK
Apr 29 22:49:00 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: Sent DUA.rsp for DUA fd00:7d03:7d03:7d03:15e7:e396:34d6:4d81, status 0 OK
```

It's unclear why PBBR re-registers. Checked the code but couldn't find any cause that may cause PBBR to reregister its DUA.

Corresponding PCAP: [test_ndproxy_9_merged.pcap.zip](https://github.com/openthread/openthread/files/6438883/test_ndproxy_9_merged.pcap.zip)

@librasungirl

simonlingoogle, 2021-05-07:
[test_ndproxy_9.log](https://github.com/openthread/openthread/files/6438897/test_ndproxy_9.log)

librasungirl, 2021-05-07:
From the log
```
Apr 29 22:48:30 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: PBBR state: Added
Apr 29 22:48:30 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: Rloc16: 0xF400, seqno: 252, delay: 1200, timeout 3600
...
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-N-DATA--: Sent server data notification
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: Add BBR Service: seqno (252), delay (5s), timeout (300s), OK
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: Set BBR Service: seqno (252), delay (5s), timeout (300s), OK
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-CORE----: Notifier: StateChanged (0x04000000) [BbrLocal]
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-N-DATA--: Received network data registration
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-N-DATA--: Allocated Service ID = 1
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-N-DATA--: Sent network data registration acknowledgment
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-CORE----: Notifier: StateChanged (0x00000200) [NetData]
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-MLE-----: Send Data Response (ff02:0:0:0:0:0:0:1)
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: PBBR state: Refreshed
Apr 29 22:48:33 05d19bedf796 otbr-agent[104]: [INFO]-BBR-----: Rloc16: 0xF400, seqno: 252, delay: 5, timeout 300
```
We can see thatThe reregistration delay in default bbr dataset (1200s) takes effect indeed, which might be the reason that we see re-registration from PBBR some time later.

I would say that the change in https://github.com/openthread/ot-br-posix/pull/724 (which enables bbr functionality by default) helps to expose one issue in existing dua implementation. When sending DUA.req, we should also reset the delay as in
https://github.com/openthread/openthread/commit/3fd53670cb0386ade3a93ee47976f8dd94a8fd47#diff-f22621e2122bf5e36a030cbcbfbe861983e321e66d3e38921632626b9eb3b47aR527

The reason why it is not occur in previous regressions, guess it might also be related with https://github.com/openthread/ot-br-posix/pull/724. The original design for the tests is to override the default bbr dataset with new configuration before explicitly enable bbr functionality in script, the new reregistration delay 5s is so short that it may decrease to 0 when `mDelay.mFields.mRegistrationDelay` becomes 0.

With https://github.com/openthread/ot-br-posix/pull/724, I think the test scripts should be updated to configure new bbr configuration before start bbr, and remove the calling of `enable_backbone_router()`. @simonlingoogle would you please revisit and update the scripts?

@simonlingoogle additionally how about enabling `DEGB` level log for otbr?  Thus more information could be provided for issue analysis.

simonlingoogle, 2021-05-08:
> @simonlingoogle additionally how about enabling `DEBG` level log for otbr? Thus more information could be provided for issue analysis.

I think DEBG is not enabled because of `otbr-agent` command argument. We need to change the log level in `/etc/default/otbr-agent`.

simonlingoogle, 2021-05-08:
> With [openthread/ot-br-posix#724](https://github.com/openthread/ot-br-posix/pull/724), I think the test scripts should be updated to configure new bbr configuration before start bbr, and remove the calling of `enable_backbone_router()`. @simonlingoogle would you please revisit and update the scripts?

I prefer slightly to turn off `Backbone Router` on BBRs before start() in tests, for these reasons:
- We have quite a tradition of calling `enable_backbone_router()` to enable `Backbone Router` in tests. It seems better to keep this tradition. 
- The BBRs in virtual time simulation (without using OTBR Docker) do not have `Backbone Router` eanbled by deafult because it's not OTBR. I think it's better to keep these two cases consistent by always turn off `Backbon Router` for BBRs.

Thoughts? @librasungirl

librasungirl, 2021-05-08:
> > With [openthread/ot-br-posix#724](https://github.com/openthread/ot-br-posix/pull/724), I think the test scripts should be updated to configure new bbr configuration before start bbr, and remove the calling of `enable_backbone_router()`. @simonlingoogle would you please revisit and update the scripts?
> 
> I prefer slightly to turn off `Backbone Router` on BBRs before start() in tests, for these reasons:
> 
> * We have a tradition of calling `enable_backbone_router()` to enable `Backbone Router` in tests. It seems better to keep this tradition since everyone is familiar with doing so.

We had this convention because that backbone router is disabled by default at that time. However now along with openthread/ot-br-posix#724 which enables BBR functionality by default, we should update the convention to reflect the current situation accordingly, and show the normal usage in case the community may refer to the test script.

> * The BBRs in virtual time simulation (without using OTBR Docker) do not have `Backbone Router` eanbled by default because it's not OTBR. I think it's better to keep these two cases consistent by always turn off `Backbon Router` for BBRs.
> 
> Thoughts? @librasungirl

Thanks for raising the discussion on BBR simulation node without backbone functionality. `is_bbr` was introduced for functional tests when we don't have complete OTBR backbone router solution in early development stage. Given that we don't have such limitation now, I don't figure out yet  whether or not we should keep such simulation device for the long term, like should we consider deprecating the tests with such device if we have complete Test Cases defined in TG's TP someday?

To keep consistent with OTBR docker device,  can we check if it is `is_bbr` simulation thread device (`is_bbr is true and is_otbr is false`) and enable_backbone_router() if it is in start() API
https://github.com/openthread/openthread/blob/8dc9a53cc7530495023e10949cbafb4c4d18ae76/tests/scripts/thread-cert/node.py#L711-L713

Thoughts?

librasungirl, 2021-05-08:
> > Thanks for raising the discussion on BBR simulation node without backbone functionality. `is_bbr` was introduced for functional tests when we don't have complete OTBR backbone router solution in early development stage. Given that we don't have such limitation now, I don't figure out yet whether or not we should keep such simulation device for the long term, like should we consider deprecating the tests with such device if we have complete Test Cases defined in TG's TP someday?
> 
> I tend to keep them because they are still useful for testing the Thread core part of Backbone Router features (e.g. DUA registration). They are also convenient because they can use `Virtual Time simulation`.

Agree that it has advantages in virtual time.  However on the other hand, these functionalities covered in these tests should be well covered in 1.2 test cases defined in 1.2 Test Plan. And when 1.2 test cases are completely available, and becomes mandatory in GitHub Action, what's the point of keeping these tests?

> 
> > To keep consistent with OTBR docker device, can we check if it is `is_bbr` simulation thread device (`is_bbr is true and is_otbr is false`) and enable_backbone_router() if it is in start() API
> 
> Either `enable Backbone Router by default` or `disable Backbone Router by default` works for me. I will submit a PR to make `Backbone Router` enabled on all BBRs by default, unless configured otherwise.

Thanks. Looking forward to it.

jwhui, 2022-03-01:
Have not seen this issue recently. Closing.

closed 2022-03-01 by jwhui