openthread/openthread #6546: [border router] total SRP server entries on a network should be limited by BR clients

ndyck14, 2021-04-30:
Unicast SRP server entries take up 22 bytes in the network data and anycast entries take 2 bytes. Both should be limited for cases where users may have many BRs.

Suggested limits:
unicast: 2 per network
anycast: 5 per network

Requirements:

1. Servers must observe the network data and if the limit for SRP servers has been reached, they must not add their own entries
2. If the limit has been exceeded, BRs must remove their own entries if their address is numerically the highest
3. If a server is inactive due to the limit, it should observe the network data and add their own entry if the number of srp servers drops below the limit

ndyck14, 2021-04-30:
We can look to take up this work. Feedback on the description is welcome

abtink, 2021-04-30:
Thanks @ndyck14 for submitting this and also offering to help with this.

Yes, we do indeed need such a feature and behavior.

I have actually already started on such a feature as a follow up to PR https://github.com/openthread/openthread/pull/6501.

We had an experimental implementation of a very similar behavior for on-mesh prefix entries in network data from years ago (it was intended for an internal project).

I expect to have a PR on this by end of next week or so.

abtink, 2021-05-07:
Just submitted PR #6587.

gabekassel, 2021-06-03:
@abtink was this issue resolved in other PRs than the one linked? I recalled you were copying SRP work between PRs as you built it up.

jwhui, 2021-06-03:
> @abtink was this issue resolved in other PRs than the one linked? I recalled you were copying SRP work between PRs as you built it up.

I don't think so - #6587 directly addresses this issue.

abtink, 2021-06-03:
@gabekassel the PR for this is the same one as before PR #6587 (it is open but I think/hope we can merge it soon).

- We added support for SRP/DNS service entries first (which is this issue). 
- Later I added "on-mesh/external route" prefix support on top of it as well which treated each prefix entry as its own (managed whether to publish it or not). 
- More recently and upon further discussions and considerations, we decided to change how we want to manage the prefix entries and treat them all as a group (the idea is on a device we want all prefix related entries to be published together or publish none of them). With this, I removed the prefix related code from this PR (back to supportng SRP/DNS service only).

ndyck14, 2021-07-16:
@abtink where is this at? Looks like PR was declined?

jwhui, 2021-07-16:
> @abtink where is this at? Looks like PR was declined?

@ndyck14 , the current PR is #6768. It is still going through review and waiting for approvals. For further context, see https://github.com/openthread/openthread/pull/6768#issue-678589067.

abtink, 2021-07-16:
Thanks @jwhui. 

- Yes. The above PR replaces the previous one and contains the full implementation (includes support DNS/SRP service  and on-mesh prefixes and external route prefixes). 
- Main thing has been the change to how domain ID is being managed by Thread (leader) which help simplify how the BRs manage such prefix entries and also made the use of `Publisher` more straigth-forward (more details in https://github.com/openthread/openthread/pull/6587#issuecomment-869232331)

closed 2021-08-11 by jwhui