openthread/openthread #6549: ECDSA build error on EFR32 (BRD4166A) platform

tutulala, 2021-05-02:
**Describe the bug** A clear and concise description of what the bug is.

While building SRP_CLIENT example with nrf52840 platform (for USB dongle) when through and work as expected with following flags:
```
make -f examples/Makefile-nrf52840 USB=1 \
BOOTLOADER=USB \
SRP_CLIENT=1    \
ECDSA=1 \
THREAD_VERSION=1.2 \
EXTERNAL_MBEDTLS=nordicsemi-mbedtls \
```

The same build is broken on EFR32 (BRD4166A) platform with following flags:
```
make -f examples/Makefile-efr32mg12 BOARD=BRD4166A \
SRP_CLIENT=1 \
ECDSA=1 \
THREAD_VERSION=1.2 \
EXTERNAL_MBEDTLS=silabs-mbedtls
```
And it ends up in following error:

```
/home/host/openthread/examples/../src/core/crypto/ecdsa.cpp: In member function 'ot::Error ot::Crypto::Ecdsa::P256::KeyPair::Generate()':
/home/host/openthread/examples/../src/core/crypto/ecdsa.cpp:67:11: error: 'mbedtls_pk_write_key_der' was not declared in this scope; did you mean 'mbedtls_ecp_write_key'?
   67 |     ret = mbedtls_pk_write_key_der(&pk, mDerBytes, sizeof(mDerBytes));
      |           ^~~~~~~~~~~~~~~~~~~~~~~~
      |           mbedtls_ecp_write_key
  CXX      mac/libopenthread_mtd_a-channel_mask.o
/home/host/openthread/examples/../src/core/crypto/ecdsa.cpp: In member function 'ot::Error ot::Crypto::Ecdsa::P256::KeyPair::Sign(const ot::Crypto::Sha256::Hash&, ot::Crypto::Ecdsa::P256::Signature&) const':
/home/host/openthread/examples/../src/core/crypto/ecdsa.cpp:138:9: error: 'mbedtls_ecdsa_sign_det' was not declared in this scope; did you mean  mbedtls_ecdsa_sign'?
  138 |         mbedtls_ecdsa_sign_det(&ecdsa.grp, &r, &s, &ecdsa.d, aHash.GetBytes(), Sha256::Hash::kSize, MBEDTLS_MD_SHA256);
      |         ^~~~~~~~~~~~~~~~~~~~~~
      |         mbedtls_ecdsa_sign
make[4]: *** [Makefile:6785: crypto/libopenthread_mtd_a-ecdsa.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:434: all-recursive] Error 1
make[2]: *** [Makefile:586: all-recursive] Error 1
make[1]: *** [Makefile:512: all] Error 2
make: *** [examples/Makefile-efr32mg12:296: do-build-cortex-m4] Error 2
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id

2. IEEE 802.15.4 hardware platform
EFR32 Thunder sense 2 board (BRD4166A).
3. Build steps
Just build SRP_CLIENT example for EFR32:
```
make -f examples/Makefile-efr32mg12 BOARD=BRD4166A \
SRP_CLIENT=1 \
ECDSA=1 \
THREAD_VERSION=1.2 \
EXTERNAL_MBEDTLS=silabs-mbedtls
```

4. Network topology
N/A

**Expected behavior** A clear and concise description of what you expected to happen.
Build should when through as it does on nrf52840 platform.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

tutulala, 2021-05-02:
It seems at least 4 defines are missed:

```
#define MBEDTLS_PK_WRITE_C
#define MBEDTLS_ECDSA_C
#define MBEDTLS_ECDSA_DETERMINISTIC
#define MBEDTLS_HMAC_DRBG_C
```

For cmake build script (when build with cmake):
This source ${SILABS_MBEDTLS_DIR}/library/pkwrite.c need to be added in the source list of <root>/third_party/silabs/cmake/mbedtls.cmake.

thanglenguyen, 2021-05-03:
Created a [pull request](https://github.com/openthread/ot-efr32/pull/41) on platform specific repo openthread/ot-efr32.

romacdon, 2021-05-03:
I assigned this @thanglenguyen since they created a PR against it - https://github.com/openthread/ot-efr32/pull/41

closed 2021-05-03 by jwhui