openthread/openthread #6568: [github-actions] test_multi_border_routers.py fails intermittently

jwhui, 2021-05-06:
[logs_8086.zip](https://github.com/openthread/openthread/files/6431530/logs_8086.zip)
[thread-1-2-backbone-results.zip](https://github.com/openthread/openthread/files/6431531/thread-1-2-backbone-results.zip)

wgtdkp, 2021-05-06:
The error is:
```txt
Packets: 23 May  6 03:01:46 77c3bf212000 otbr-agent[111]: [CRIT]-PLAT----: HandleRcpTimeout() at ../../third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2187: RadioSpinelNoResponse
connect session failed: No such file or directory
```

It doesn't look like an issue with the test itself.

jwhui, 2021-05-21:
[logs_78722.zip](https://github.com/openthread/openthread/files/6519547/logs_78722.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6519548/thread-border-router-results.zip)

jwhui, 2021-05-26:
[logs_79854.zip](https://github.com/openthread/openthread/files/6549868/logs_79854.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6549870/thread-border-router-results.zip)

wgtdkp, 2021-05-29:
The log indicates that there are no OMR address on the BR even when there is one OMR prefix. This does sound reasonable.
For the test script is parsing command output and matching the address and prefix, suspect it has something to do with https://github.com/openthread/openthread/issues/6665.
Will recheck if the issue disappears when https://github.com/openthread/openthread/issues/6665 is fixed.

jwhui, 2021-10-15:
[logs_98618.zip](https://github.com/openthread/openthread/files/7356370/logs_98618.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/7356371/thread-border-router-results.zip)

jwhui, 2022-09-13:
Closing stale issue.

closed 2022-09-13 by jwhui