openthread/openthread #6575: Cyclic deamon quit at HandleRcpTimeout

Endag, 2021-05-06:
OTBR daemon repeatedly quits (and get restarted by systemd to only quit again) complaining about timeout of communication with RCP.

**To Reproduce** Information to reproduce the behavior, including:

**OTBR build:**
Git commit: 8f23e77966a36ed67eb99b9be14e6189806977b6
Cmake build options: -D BUILD_TESTING=OFF -D OTBR_BORDER_ROUTING=ON -D OTBR_WEB=ON -D OTBR_REST=ON -D THREAD_VERSION=1.2

Build itself made by yocto system for custom board, can provide full build logs if neccessary

**RCP build**
Git commit: 8467a3d5223c274b65a71080f6aad94dc182a9bb
Build command:  make -f examples/Makefile-nrf52840 USB=1 BOOTLOADER=1 THREAD_VERSION=1.2 COMMISSIONER=1
Platform: nrf52840

**Network topology**
RCP connected to otbr's board via usb.
No stable network.

**Expected behavior** OTBR works reliably without constant restarts

**Console/log output** [full log here](https://gist.github.com/Endag/bd3dd74dd9adb41b9c1c59dd16dbca09)

**Additional context** After system boot it seem to be "stable" (not restarting) until you try to do something (form/commission/get status) via ot-ctl

jwhui, 2021-05-06:
@Endag , thanks for raising this issue.

For the RCP, are you using the [nRF52840 Dongle](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle) or your own custom hardware?

Are you able to test with the reference hardware (Raspberry Pi 3B+ or 4B and a nRF52840 Dongle)? If so, do you see similar behavior?

Endag, 2021-05-07:
> For the RCP, are you using the [nRF52840 Dongle](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle) or your own custom hardware?

It was custom hw, tested with  PCA10056 DK and same RCP fw, it seems to work fine, so it must be something in rcp's platform part of code.

Mine RCP-board version is mainly breakdown board for interfaces and  radio/usb on standart pins, so it was working fine when it was flashed with NCP firmware from Nordic SDK(so i didn't think to check rcp with other hw). 

Any suggestions where to check platform code platform code for pollible issue source?

jwhui, 2021-05-07:
@Endag , the logs show that the host is not receiving an indication from the RCP that the transmit operation completed successfully and timed out.

I am not able to reproduce this issue using a Raspberry Pi 4B and a nRF52840 Dongle. Can you confirm whether the RCP firmware you built experiences the same issue using the reference hardware (Raspberry Pi 3B+ or 4B and a Nordic board)?

Endag, 2021-05-07:
As i said, same firmware works fine when i using standart Nordic PCA10056-DK board as RCP.

Only reason i can come up now for such behavior is that RCP firmware uses some pins that cause  some errors in hardware, but for now i cannot  figure which.
Pin used by radio driver's FEM are unused at my board.

jwhui, 2021-05-07:
> As i said, same firmware works fine when i using standart Nordic PCA10056-DK board as RCP.

Thanks for clarifying. You had also mentioned using firmware from the Nordic SDK, so I wasn't sure which you were talking about.

> Only reason i can come up now for such behavior is that RCP firmware uses some pins that cause some errors in hardware, but for now i cannot figure which.
> Pin used by radio driver's FEM are unused at my board.

Maybe @LuDuda @MarekPorwisz @pszkotak can help provide some guidance?

lmaciejonczyk, 2021-05-12:
Hi @Endag,
I'm not an expert in a domain you have an issue in, but maybe this pin configuration is worth of checking on your breakdown board:
https://github.com/openthread/ot-nrf528xx/blob/af5d0331cd48a24f4d33c8b888ac659396f83a79/third_party/NordicSemiconductor/config/app_config.h#L139
More information about `app_config.h` file and its genesis of it you can find here: https://github.com/openthread/openthread/pull/4511

jwhui, 2021-06-21:
Closing stale issue.

closed 2021-06-21 by jwhui