openthread/openthread #6583: [log] bad string/output in logs

abtink, 2021-05-06:
I have noticed recently some bad/weird formats in some of the log lines (e.g., the `Send Advertisement` below).
```
[INFO]-MLE-----: Receive Link Accept and Request (fe80:0:0:0:30e0:f941:5b85:5122,0x9000)
"?Oմ*)-MLE-----: Send Link Accept (fe80:0:0:0:30e0:f941:5b85:5122oZ~??.]{P
[INFO]-MAC-----: Sent IPv6 UDP msg, len:107, chksum:5906, to:32e0f9415b855122, sec:no, prio:net
[INFO]-MAC-----:     src:[fe80:0:0:0:f0ef:3bc:197e:3434]:19788
[INFO]-MAC-----:     dst:[fe80:0:0:0:30e0:f941:5b85:5122]:19788
[INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1b?
                                                         j?QZP?????*)
[INFO]-MAC-----: Sent IPv6 UDP msg, len:91, chksum:b113, to:0xffff, sec:no, prio:net
[INFO]-MAC-----:     src:[fe80:0:0:0:f0ef:3bc:197e:3434]:19788
[INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
[INFO]-MAC-----: Received IPv6 UDP msg, len:91, chksum:55c5, from:32e0f9415b855122, sec:no, prio:net, rss:-20.0
```

Adding this issue to investigate this further.

abtink, 2021-05-06:
I think I found the cause of this issue. It seems that it was introduced from PR https://github.com/openthread/openthread/pull/6564 (adding `StringWriter`).

Here is, for example, the method that logs MLE message (e.g., `Send Advertisement`).
```c++
void Mle::Log(MessageAction aAction, MessageType aType, const Ip6::Address &aAddress, uint16_t aRloc)
{
    enum : uint8_t
    {
        kRlocStringSize = 17,
    };

    String<kRlocStringSize> rlocString;

    if (aRloc != Mac::kShortAddrInvalid)
    {
        StringWriter(rlocString).Append(",0x%04x", aRloc);
    }

    otLogInfoMle("%s %s%s (%s%s)", MessageActionToString(aAction), MessageTypeToString(aType),
                 MessageTypeActionToSuffixString(aType, aAction), aAddress.ToString().AsCString(),
                 rlocString.AsCString());
}
```

The issue is that this code assumed that `String<kRlocStringSize> rlocString` will start as an empty string, but now we don't have a constructor in `String` anymore and initialize it only from constructor of `StringWriter`. So in the case that `aRloc` is `kShortAddrInvalid` we never call `StringWriter(rlocString)` and the string remains uninitialized.

Before noticing this, I submitted PR https://github.com/openthread/openthread/pull/6582 to add the constructor for `String` (to init as empty) but it impacts code size.

Here are the options I see:
1. We can add a constructor back in `String` to init it as empty. This will be safer but adds code size.
2. We can handle each case (like above) and make sure `String` is always used with a `StringWriter`. The downside I see  is that I feel it can be easy to make a mistake with this approach (and then we can end up using uninitialized string). 

I lean towards option (1) adding the constructor.

Thoughts? @bukepo .

abtink, 2021-05-06:
Found a few other cases that same issue can happen, e.g., in `link_quality.cpp`:
```
RssAverager::InfoString RssAverager::ToString(void) const
{
    InfoString string;

    VerifyOrExit(mCount != 0);
    StringWriter(string).Append("%d.%s", -(mAverage >> kPrecisionBitShift),
                                kDigitsString[mAverage & kPrecisionBitMask]);

exit:
    return string;
}
```

Also in `RadioSelector::Log()` method.

abtink, 2021-05-07:
Submitted PR https://github.com/openthread/openthread/pull/6584 to fix this.  This can impact beyond just getting bad log (can potentially cause crash) so I think we want to address it quickly.

The idea in this PR is that any `String` variable definition is always immediately followed by a `StringWriter writer(string)` definition which then initializes the string buffer (sets it as empty). I went through and changed sall the current uses of `String`. I suggest we ensure to use (maybe even enforce) this pattern in any future use of `String`.

bukepo, 2021-05-07:
Thank you for the quick fix! I thought the static analyzers can find such indeterminate values. Let's see if we can find some enhancements to the GitHub action checks.

simonlingoogle, 2021-05-07:
> 1. We can add a constructor back in `String` to init it as empty. This will be safer but adds code size.
> 2. We can handle each case (like above) and make sure `String` is always used with a [...]

Why didn't we go with solution 1. I think option 1 is better so that `StringWriter` is just responsible for appending characters to the string. It does not care if the original string is empty or not.

abtink, 2021-05-07:
> Why didn't we go with solution 1. I think option 1 is better so that StringWriter is just responsible for appending characters to the string. It does not care if the original string is empty or not.

I do like this as well sine it is a safer model, but did not go for it for now due to its code size impact. Adding a constructor to clear the `String` adds up to ~110 bytes (please see https://github.com/openthread/openthread/pull/6582). It somewhat defeats one of the main benefits of `StringWriter` (which is/was to reduce code size).  For now, I think would be good to fix this quickly and then  later we can consider different options.

closed 2021-05-07 by jwhui