openthread/openthread #6586: Potential vulnerability in remove - re-add service pattern.

kkasperczyk-no, 2021-05-07:
**Problem**
Some time ago we discussed with @abtink @LuDuda @Damian-Nordic and @jwhui the problem of missing service update feature and we concluded it that this is how SRP protocol works like, so in case of necessity to update specific service record it should be removed and re-added once again. 

I introduced that solution in the Project CHIP, but @msandstedt noticed (https://github.com/project-chip/connectedhomeip/pull/6539#issuecomment-833593097) that there is potential security problem with this pattern and bad actor may try to steal removed service name and add it with own key, before the SRP client will manage to do that again. I'm not able to judge if it is a real threat, but seems to be worth considering. @abtink @jwhui could you comment on that?

**Alternatives**
I think that risk could be reduced by sending remove and following add requests in one SRP Update message thanks to the  OPENTHREAD_CONFIG_SRP_CLIENT_UPDATE_TX_DELAY feature (https://github.com/openthread/openthread/blob/main/src/core/config/srp_client.h#L162), however it doesn't seem to solve problem in 100%, as for some reason OpenThread thread could be blocked by some other application thread and that would result in TX_DELAY expire and sending two separate SRP Updates.

**Proposed solution**
I wonder if it would be reasonable to implement some API allowing to force sending few requests in one SRP Update message. I imagine some `StartTransaction` method that would start gathering requests and `StopTransaction` that would stop it and send on SRP Update filled with whole gathered data at once.

wgtdkp, 2021-05-07:
> Some time ago we discussed with @abtink @LuDuda @Damian-Nordic and @jwhui the problem of missing service update feature and we concluded it that this is how SRP protocol works like, so in case of necessity to update specific service record it should be removed and re-added once again.

Not sure if I understand the problem correctly, but if your requirement is to update service RRs like TXT entries or port number in SRV RR, I think SRP supports this well. The client can always re-new the SRP service with different RRs.

I see you PR in the CHIP repo https://github.com/project-chip/connectedhomeip/pull/6539. The problem is that you are trying to add two different services with the same instance and service name and I believe this is not CHIP want to do: the device want to update the RRs associated with the same service (identified by the union of `service name` and `instance name`), right?

We should reuse the SRP service entry with the same name on the CHIP end device but not always adding a new one.

kkasperczyk-no, 2021-05-07:
> Not sure if I understand the problem correctly, but if your requirement is to update service RRs like TXT entries or port number in SRV RR, I think SRP supports this well. The client can always re-new the SRP service with different RRs.

Thanks for answering on that @wgtdkp, I answered your questions in PR comment: https://github.com/project-chip/connectedhomeip/pull/6539#discussion_r628025782

wgtdkp, 2021-05-07:
https://github.com/project-chip/connectedhomeip/pull/6539#discussion_r628051535 @abtink @jwhui  thoughts?

abtink, 2021-05-07:
> there is potential security problem with this pattern and bad actor may try to steal removed service name and add it with own key, before the SRP client will manage to do that again

I don't think this would be possible or an issue. The SRP has the notion of "key lease". The idea is that even when an entry is removed, the SRP server will retain the instance name along with the key used when it was registered for "key lease" interval. This is to disallow any other device to claim the same name. The original device can come back and register/claim the same instance name again (using the same key which device itself retains).

abtink, 2021-05-07:
If I understand correctly the use-case is to change some parameter in an already existing service (e.g., change port number, priority, or TXT info, etc) for the same service instance name.

In the SRP client OT API design we require that the content in an `otSrpClientService` struct to persist and not be changed by the caller after it is given to an OT API and it can be reclaimed/reused only after the callback indicating that it was removed. I will get back to this point below (whether we can relax this and what are its consequences).

As @kkasperczyk-no mentioned, the update to an existing service can be done by removing the existing entry and adding a new one with same name but updated info.
- We can call `otSrpClientRemoveService()` on the existing `otSrpClientService` 
   - Note that even after this call, the `otSrpClientService` entry content should not be changed and only after callback that remove was done, it can be reclaimed and reused.
- Then immediately follow by `otSrpClientAddService()` with a new/different `otSrpClientService` entry containing same instance name but with new parameters.

I expect this flow to work fine and basically include the "remove and add" instructions in the same SRP Update Message.

> I think that risk could be reduced by sending remove and following add requests in one SRP Update message thanks to the OPENTHREAD_CONFIG_SRP_CLIENT_UPDATE_TX_DELAY feature, however it doesn't seem [...]

- If the two calls to `otSrpClientRemoveService()` and `otSrpClientAddService()` happen back to back then it should always go in the same SRP Update message.
   - The actual SRP message preparation and tx happens from a timer callback (after a short delay)
   - In general, we require/assume that OT public APIs (e.g., `otSrpClient{Add/Remove}Service()`) and/or the platform callbacks (e.g. timer callback) to be invoked from the same platform OS context (i.e., same RTOS task or same thread, etc). And everything is processed from `Tasklets` which keep the order (and cannot to be preempted).
   - So if we call `Remove()/Add()` back to back from same function/method, even if the timer is expired, the timer callback cannot be invoked in between the two calls.

I agree that this may be a bit inefficient in the sense that we need to use two `otSrpClientService` instances (cannot just change the existing one).

-------------------

Back to the point about allowing an existing/registered `otSrpClientService` instances to be changed by caller (after it is given to the OT):
- I agree this can help allow reusing the same entry and we can add a new OT API to "update" an existing service (not need to remove and add with two different entry instances).
- My concern is that if we relax the requirement for `otSrpClientService` to not be changed by caller, it may lead to potential incorrect use and/or tricky bugs.
   -  For example, we can have a race situation if the next layer code tries to change some fields/pointers in an already registered `otSrpClientService` struct from a different OS context (e.g., different RTOS task) while the OT core is also accessing the same content.
- We need to consider how helpful adding an "update service" OT API will be (compared to using "remove" / "add" with two entries) and the complexity it can add and think of how to make its use safer:
  - For example, we can add a new API that allows the caller to claim back the ownership of an existing `otSrpClientService` instance.
  - After calling this, the caller (next layer) can change the `otSrpClientService` content and then call a new OT API to give it back to OT and request a refresh of this service entry.
 
Thoughts?

abtink, 2021-05-09:
> - We need to consider how helpful adding an "update service" OT API will be (compared to using "remove" / "add" with two entries) and the complexity it can add and think of how to make its use [...]

Thinking more on this, I came up with an approach to allow caller (next layer code) to re-use an existing service entry to update some service paramter without relaxing our requirement for the `otSrpClientService` not to be changed while it is given to and owned by OT core (`Srp::Client` module).

The idea is we add a new `DeleteService()` (or `otSrpClientDeleteService()` as public OT API) method
- This method deletes a service from SRP client, immediately removing it from the list. 
- Unlike `RemoveService()` which sends an SRP update message to server to remove/unregister the service on server , this method deletes the service from the client list without any interaction with server. 
- On a successful call to this method, the `Callback` will NOT be called and service entry can be reclaimed and re-used by the caller immediately.
- Basically after `DeleteService()` the service is removed from clinet, but will still be present on server till it expires its lease time 

We can then use the `DeleteService()` along with a subsequent call to `AddService()` (potentially reusing the same service entry with the same service and instance names) to  update some of the parameters in an existing service in next layer code.

I have submitted PR https://github.com/openthread/openthread/pull/6598 which adds `DeleteService()` method. Another (somewhat) related PR is PR https://github.com/openthread/openthread/pull/6597 which protects against caller trying to add duplicate service entries (with the same service and instance names).

kkasperczyk-no, 2021-05-10:
@abtink thank you for explaining exactly how SRP works and dispelling any doubts on that. I'm closing this issue, as for me reported problem is no longer valid.

closed 2021-05-10 by kkasperczyk-no