openthread/openthread #6588: OpenThread And Azure IoT Hub question

McDope999, 2021-05-07:
I'm currently working on a project where we are looking into using OpenThread along with Azure IoT Hub, and I would like to hear if anyone anyone have experience/knowledge about integrating an OpenThread device with Azure IoT Hub? I'm new to both technologies, but one of the problem i'm facing is that Azure IoT SDK is supporting MQTT, which is not supported by OpenThread because of the UDP transport layer. Azure is not supporting CoAP, so I need some sort of binding/Gateway between those to. Currently I have tried with some MQTT-SN, where I have succeeded in publish/subscribing to topics on a MQTT Broker with OpenThread, but Azure IoT SDK has some different demands regarding connecting and device firmware update, which I don't think MQTT-SN will solve. Azure has an IoT Protocol Gateway, which is maybe what I need. But I would like to hear if anyone has any experience, as I can't find any information on this specific use case? Any advice, ideas and inspiration would be very much appreciated. Thanks.

jwhui, 2021-06-19:
@McDope999 , I don't have any direct experience with Azure IoT Hub. However, there are examples of MQTT over OpenThread.

- OpenThread has an example integration with FreeRTOS, LwIP, and MQTT: https://github.com/openthread/ot-rtos
- Zephyr supports both OpenThread and MQTT: https://docs.zephyrproject.org/latest/reference/networking/mqtt.html
- Nordic SDK for Thread and Zigbee includes an MQTT-SN example: https://www.nordicsemi.com/Software-and-tools/Software/nRF5-SDK-for-Thread-and-Zigbee

Hope that helps.

jwhui, 2021-08-25:
Closing stale issue.

closed 2021-08-25 by jwhui