openthread/openthread #6594: send message

anying11, 2021-05-09:
What should I do if I want to send a string from a end device to a border router through cli?

jwhui, 2021-05-09:
See [src/cli/README_UDP.md](https://github.com/openthread/openthread/blob/main/src/cli/README_UDP.md).

anying11, 2021-05-09:
Can  the ipv6 address be found  through cli ,or I have to  make up it ?

jwhui, 2021-05-09:
> Can the ipv6 address be found through cli ,or I have to make up it ?

See [src/cli/README.md#ipaddr](https://github.com/openthread/openthread/blob/main/src/cli/README.md#ipaddr)

anying11, 2021-05-09:
In the "udp send <ip> <port> <message>" partition,what does port mean?Does this mean that the port it already use or a brand new port?

simonlingoogle, 2021-05-10:
The port of `udp send` command is the destination port.

anying11, 2021-05-12:
May I ask what does the "eventTypeAlarmFired" mean?

anying11, 2021-05-19:
How can I get all the information in all of the border routers,even without the connection to Wifi?Is this possible?

jwhui, 2021-06-21:
Closing stale issue.

closed 2021-06-21 by jwhui