openthread/openthread #6595: Broken handling for retransmission decision while decryption (Authentication) of Enhanced Ack

khaled34, 2021-05-09:
In case of radio supports both OT_RADIO_TRANSMIT_SEC but doesn't support OT_FRAME_RETRIES capabilities. What if secured enhanced ack is received by a device but the  failed to be decrypted  <Authentication error>. does this require retransmission of the same packet or not? 

Based on that question and after investigating in the OT 1.2 latest code i found that if radio doesn't support MAX_FRAM_RETRIES the functionality will be on the sub_mac layer as described below:
void SubMac::HandleTransmitDone(TxFrame &aFrame, RxFrame *aAckFrame, Error aError){
....
.....

 // Determine whether to re-transmit the frame.
shouldRetx = ((aError != kErrorNone) && ShouldHandleRetries() && (mTransmitRetries < aFrame.GetMaxFrameRetries()));
mCallbacks.RecordFrameTransmitStatus(aFrame, aAckFrame, aError, mTransmitRetries, shouldRetx);
    if (shouldRetx)
    {
        mTransmitRetries++;
        aFrame.SetIsARetransmission(true);
        StartCsmaBackoff();
        ExitNow();
    }
SetState(kStateReceive);
mCallbacks.TransmitDone(aFrame, aAckFrame, aError); // >>>>
}
As I can see that handling of Enhanced ack decryption is not done in the sub_mac layer and so it isn't governing by the retransmission decision. I also saw mac::ProcessEnhAckSecurity which is used in case of mac::handleTransmitDone but it is called after max number of retransmission failure.

When radio doesn't support MAX_FRAME_RETRIES where does the decryption take place to allow retry?

jwhui, 2021-05-15:
@khaled34 , thanks for raising this question.

OpenThread does not currently implement authentication of secured Enh-Ack frames before determining whether to retransmit a given data message. This is intentional.

OpenThread does, however, implement authentication of secured Enh-Ack frames before processing any IEs or MAC payload. So any sensitive information, such as timing information for CSL or link metrics carried in IEs will be properly authenticated.

There is not much added value to authenticating the Enh-Ack before determining retransmission. Any attack that takes advantage of this fact is a form of DoS attack. The end result is not much different from an RF jamming attack. At the same time, requiring the sub-MAC to perform authentication adds significant overhead to the RCP, including needing to maintain frame counters for each neighboring device. For this reason, OpenThread does not currently perform authentication of Enh-Ack frames at the sub-MAC layer.

jwhui, 2021-06-19:
Closing stale issue.

closed 2021-06-19 by jwhui