openthread/openthread #6616: [github-actions] ot-commissioner test fails intermittently

jwhui, 2021-05-12:
[logs_76239.zip](https://github.com/openthread/openthread/files/6462852/logs_76239.zip)

jwhui, 2021-09-29:
[logs_97392.zip](https://github.com/openthread/openthread/files/7254844/logs_97392.zip)

jwhui, 2022-01-13:
[logs_110331.zip](https://github.com/openthread/openthread/files/7865642/logs_110331.zip)

jwhui, 2022-01-19:
From @wgtdkp in https://github.com/openthread/openthread/issues/7333#issuecomment-1016424285:

---

Looks like there is something broken with the new Commissioner enhancement tests. But not clear why it starts fail now, after running stably for nearly half a year:
```
2022-01-19T11:13:40.8845002Z ++ grep -q '1111111111111111": false'
2022-01-19T11:13:40.8845330Z ++ send_command_to_commissioner 'sessionid --nwk other' 'DEAD00BEEF00CAFE": \d*'
2022-01-19T11:13:40.8845621Z ++ local 'command=sessionid --nwk other'
2022-01-19T11:13:40.8845821Z ++ local expect_error=
2022-01-19T11:13:40.8846184Z ++ '[' 2 == 2 ']'
2022-01-19T11:13:40.8846423Z ++ expect_error='DEAD00BEEF00CAFE": \d*'
2022-01-19T11:13:40.8846597Z ++ local result
2022-01-19T11:13:40.8846890Z +++ /usr/local/bin/commissioner_ctl.py execute 'sessionid --nwk other'
2022-01-19T11:13:40.8847148Z ++ result='[32mnull
2022-01-19T11:13:40.8847325Z [done][0m
2022-01-19T11:13:40.8847479Z thread2'
2022-01-19T11:13:40.8847654Z ++ echo '[32mnull
2022-01-19T11:13:40.8847842Z [done][0m
2022-01-19T11:13:40.8847980Z thread2'
2022-01-19T11:13:40.8848160Z [32mnull
2022-01-19T11:13:40.8848329Z [done][0m
2022-01-19T11:13:40.8848483Z thread2
2022-01-19T11:13:40.8848670Z ++ '[' -n '[32mnull
2022-01-19T11:13:40.8848842Z [done][0m
2022-01-19T11:13:40.8849026Z thread2' ']'
2022-01-19T11:13:40.8849239Z ++ '[' -z 'DEAD00BEEF00CAFE": \d*' ']'
2022-01-19T11:13:40.8849455Z ++ echo '[32mnull
2022-01-19T11:13:40.8849805Z [done][0m
2022-01-19T11:13:40.8849964Z thread2'
2022-01-19T11:13:40.8850185Z ++ grep -q 'DEAD00BEEF00CAFE": \d*'
2022-01-19T11:13:40.8850387Z ++ return 1
```

Will take a closer look tomorrow.

simonlingoogle, 2022-01-27:
another fail 
[logs_112214.zip](https://github.com/openthread/openthread/files/7946967/logs_112214.zip)

jwhui, 2022-03-10:
https://github.com/openthread/openthread/runs/5497703435?check_suite_focus=true

jwhui, 2022-03-25:
https://github.com/openthread/openthread/runs/5695191508?check_suite_focus=true

jwhui, 2022-08-23:
https://github.com/openthread/openthread/runs/7983414485?check_suite_focus=true

jwhui, 2025-03-11:
Stale issue.

closed 2025-03-11 by jwhui