openthread/openthread #6623: nRF52840 DK halt since #6243

simonlingoogle, 2021-05-14:
**Describe the bug** 

nRF52840 DK halts soon after `thread start`.

**To Reproduce**

```bash
git checkout db1b980e57ada62f6ef2002fcaddff37d55a356d
git clean -xfd
./bootstrap
make -f examples/Makefile-nrf52840 BORDER_ROUTER=1 SERVICE=1 REFERENCE_DEVICE=1 COMMISSIONER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 JOINER=1 MAC_FILTER=1 THREAD_VERSION=1.2 DUA=1 MLR=1 CSL_RECEIVER=1 FULL_LOGS=1
arm-none-eabi-objcopy -O ihex ./output/nrf52840/bin/ot-cli-ftd ot-cli-ftd.hex
nrfjprog -f nrf52 --chiperase --program ot-cli-ftd.hex --reset

> ifconfig up
Done
> thread start
Done
... Halt
```

1. Git commit id: db1b980e57ada62f6ef2002fcaddff37d55a356d
2. IEEE 802.15.4 hardware platform: nRF52840 DK

**Expected behavior** 

Halt not.

**Additional context** 
- Any commit before db1b980e57ada62f6ef2002fcaddff37d55a356d does not halt.
- nRF52840 Dongle seems not halt.
- OT LCI simulation do not halt

konradderda, 2021-05-15:
Hello, 
The issue you have reported is related to the implementation of the 'cli_uart.cpp' file. The static function named `CliUartOutput` causes the faulty behavior because of the following line:
`error = otPlatUartFlush();`

Since nRF52 do not implement `otPlatUartFlush` function they return `OT_ERROR_NOT_IMPLEMENTED` error code, this causes execution of `else` block which indicates an error with a log:
`otLogWarnPlat("Failed to output CLI: %s", otThreadErrorToString(error));`

This causes the recursion with the same failure and exhausts the stack (I did not check exactly but debugger shows a few dozen stack frames). Another problem that I've spotted in this implementation is that this function tries to transmit bytes synchronously with `while (sTxLength != 0)`, but the nRF52 UART driver works asynchronously, so even if we handle lack of otPlatUartFlush() this code most probably will not work properly. The bytes will be transmitted but `sTxLength` will not reach 0 value, because successful transmission is reported by UART driver from `nrf5UartProcess()` function that is handled from main loop context.

konradderda, 2021-05-15:
I have proposed a fix: https://github.com/openthread/ot-nrf528xx/pull/51

simonlingoogle, 2021-05-17:
> I have proposed a fix: [openthread/ot-nrf528xx#51](https://github.com/openthread/ot-nrf528xx/pull/51)

Thanks for the fix.

BTW, I verified that the fix works. @jwhui

closed 2021-05-17 by jwhui