openthread/openthread #6624: request to create a repository to add new platform support

yiyang-tian, 2021-05-14:
I want to add telink b91 support for openthread , so would you like to create a repository for me ,and the repository name should be ot-b91, thanks.

jwhui, 2021-05-14:
@yiyang-tian , thank you for contributing a new platform to the OpenThread project!

In case you have not seen yet, we have a porting guide available at https://openthread.io/guides/porting

Please stage your example in your own GitHub repo. When you are ready to move that into the OpenThread organization, let us know and we can clone the repo into the OpenThread organization.

yiyang-tian, 2021-05-24:
@jwhui I have finished the porting works , please refer the link https://github.com/yiyang-tian/ot-b91, any problems please let me know as soon as possible ,thanks.

jwhui, 2021-05-25:
@yiyang-tian , thanks for putting this together! I took a brief look and it looks good overall.

Can you also add the GitHub Actions continuous integration checks? You can find an example in https://github.com/openthread/ot-cc2538/blob/main/.github/workflows/build.yml

yiyang-tian, 2021-06-08:
@jwhui I have added the GitHub Actions continuous integration checks, you can check it again and please let me know if there is other problems.

jwhui, 2021-06-09:
@yiyang-tian , looks good overall. I have created the repo here: https://github.com/openthread/ot-b91

If everything looks good to you, I think we can close this issue.

yiyang-tian, 2021-06-10:
@jwhui Thanks，I will close this issue.

closed 2021-06-10 by yiyang-tian

lecndav, 2022-03-24:
Hi @yiyang-tian, can you please provide the start up script and the initialisation function which is compiled into libb91-drivers.a?