openthread/openthread #6643: [Proposal] Update the logging APIs

zhanglongxia, 2021-05-18:
**Is your feature request related to a problem? Please describe.** 

The [logging.hpp](https://github.com/openthread/openthread/blob/main/src/core/common/logging.hpp) is about 2600 lines now. When the developer adds a new module to OpenThread, new log APIs will be added into the file logging.hpp. Which makes the file bigger and bigger. 

**Describe the solution you'd like**

Add several generic log APIs like [ot-br-posix logging.h](https://github.com/openthread/ot-br-posix/blob/main/src/common/logging.hpp#L222). The logging prefix is defined by a macro `OT_LOG_TAG`. Each module can define its own macro `OT_LOG_TAG`. This method is very flexible for all modules. And we don't need to update the logging.hpp file anymore. The usage of the `OT_LOG_TAG` can refer to the [OTBR_LOG_TAG](https://github.com/openthread/ot-br-posix/blob/main/src/agent/advertising_proxy.cpp#L34).

Any comments? @jwhui @abtink @bukepo @librasungirl @simonlingoogle @wgtdkp

abtink, 2021-05-18:
This sounds like a great idea. I agree fully that `logging.hpp` is large and repeating a lot of same patterns. 

If I can suggest another thing that we can try to do with this change: It would be good to separate the logging for platform layer. We have platform as a log region `OT_LOG_REGION_PLATFORM` and provide the `otLogInfoPlat()` (and other flavors for different log level) for use by platform layer code. This means the platform code needs to include "logging.hpp" (which is really a core internal header file). 

Another thing to consider: We have a mode to do all logging through macros (even have the `otPlatLog()` itself being a macro) under `OPENTHREAD_CONFIG_LOG_DEFINE_AS_MACRO_ONLY` config. This was added earlier for use in an internal project/product and intended for special platform requirements where the logging needs to be defined a macro to enable certain platform feature (e.g., for log tokenization on a resource-limited device). It limits certain optimizations/changes we can do related to logging. We can consider whether it may be worth removing this config/behavior.  Thoughts?

bukepo, 2021-05-19:
Agree with @zhanglongxia and @abtink . Also I hope there are two styles logging API `otLogInfo()` and `otLogInfoV()` like `syslog()` and `vsyslog()`.

simonlingoogle, 2021-05-19:
SGTM as long as no addition to code size. :+1:

zhanglongxia, 2021-05-19:
@abtink Thanks for your great suggestions.

For the platform logging API `otLogxxxPlat`, we may define a generic logging API `otLog(otLogLevel  aLogLevel, const char *aRegionPrefix, const char *aFormat, ...)` in the file `openthread/logging.h`. The Platform layer code can call this generic API to output log.

For the macro `OPENTHREAD_CONFIG_LOG_DEFINE_AS_MACRO_ONLY `, I checked our internal projects. No one is using it. So we may remove it from OpenThread.

abtink, 2021-05-19:
> For the platform logging API otLogxxxPlat, we may define a generic logging API otLog(otLogLevel aLogLevel, const char *aRegionPrefix, const char *aFormat, ...) in the file openthread/logging.h. The Platform layer code can call this generic API to output log.

This can work, but then all platform layer implementations that use the current definitions need to be updated. May be worth it to keep same `otLogxxxPlat()` names (but maybe defining them in a public header `openthread/logging.h`). This way only the header file name need to be changed in the platform layer code.

> For the macro OPENTHREAD_CONFIG_LOG_DEFINE_AS_MACRO_ONLY , I checked our internal projects. No one is using it. So we may remove it from OpenThread.

jwhui, 2023-03-13:
Resolved by #7268 and #7385

closed 2023-03-13 by jwhui