openthread/openthread #6648: Why use IEEE 802.15.4-2006?

purvisb, 2021-05-20:
[This page](https://openthread.io/guides/porting) says openthread requires a [IEEE 802.15.4-2006](https://standards.ieee.org/standard/802_15_4-2006.html) 2.4 GHz radio. 

Why not use a newer version of 802.15.4 like [IEEE 802.15.4-2011](https://standards.ieee.org/standard/802_15_4-2011.html) or [IEEE 802.15.4z-2020](https://standards.ieee.org/standard/802_15_4z-2020.html)?

I am naïve on this subject. Please bear that in mind when responding.

jwhui, 2021-05-20:
IEEE 802.15.4-2006 provides the broadest silicon availability and compatibility with existing product designs. Thread was designed with the goal of leveraging proven technologies, economies of scale, and fast time to market.

purvisb, 2021-05-20:
@jwhui Thank you for the prompt response. That makes sense.

Any initial thoughts on the challenges that would be encountered with modifying OpenThread to use 802.15.4z-2020? Or are there no obvious obstacles (besides development time)?

jwhui, 2021-05-20:
I haven't looked into it much myself.

You may find this effort interesting: https://github.com/Decawave/mynewt-dw1000-core

purvisb, 2021-05-20:
Thank you for sharing.

closed 2021-05-20 by purvisb

markus-becker-tridonic-com, 2021-05-21:
@jwhui are you sure that IEEE 802.15.4-2006 is enough? Especially OT_RADIO_CAPS_TRANSMIT_TIMING and OT_RADIO_CAPS_TRANSMIT_SEC might require a newer version of the standard, or?

jwhui, 2021-05-21:
> @jwhui are you sure that IEEE 802.15.4-2006 is enough? Especially OT_RADIO_CAPS_TRANSMIT_TIMING and OT_RADIO_CAPS_TRANSMIT_SEC might require a newer version of the standard, or?

@markus-becker-tridonic-com , it depends on the Thread version and the features within. IIEEE 802.15.4-2015 is only required when implementing Enh-Ack or CSL as a part of Thread 1.2. Thread 1.2 routers are required to support IEEE 802.15.4-2015, but not Thread 1.2 end devices. Also Thread 1.1 devices are not required to support IEEE 802.15.4-2015.