openthread/openthread #6654: [github-actions] test_advertising_proxy.py fails intermittently

jwhui, 2021-05-21:
[logs_78728.zip](https://github.com/openthread/openthread/files/6519563/logs_78728.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6519564/thread-border-router-results.zip)

jwhui, 2021-06-04:
[logs_80611.zip](https://github.com/openthread/openthread/files/6595989/logs_80611.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6595990/thread-border-router-results.zip)

jwhui, 2021-06-10:
[logs_6285.zip](https://github.com/openthread/openthread/files/6634493/logs_6285.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6634494/thread-border-router-results.zip)

jwhui, 2021-06-24:
[logs_83067.zip](https://github.com/openthread/openthread/files/6711758/logs_83067.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6711759/thread-border-router-results.zip)

jwhui, 2021-08-09:
[logs_90670.zip](https://github.com/openthread/openthread/files/6956947/logs_90670.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/6956949/thread-border-router-results.zip)

jwhui, 2021-09-17:
[logs_96099.zip](https://github.com/openthread/openthread/files/7188035/logs_96099.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/7188036/thread-border-router-results.zip)

wgtdkp, 2021-10-20:
Just submitted https://github.com/openthread/openthread/pull/7089 to include SRP update ID in the log messages.

jwhui, 2022-01-19:
[logs_110859.zip](https://github.com/openthread/openthread/files/7893380/logs_110859.zip)
[thread-border-router-results.zip](https://github.com/openthread/openthread/files/7893381/thread-border-router-results.zip)

wgtdkp, 2022-01-19:
The failure is that the SRP server service TLV is not propagated to Network Data before timeout so that the SRP client doesn't re-register the same services after the SRP server is restarted.

```
2022-01-19T00:56:24.2261258Z ======================================================================
2022-01-19T00:56:24.2261489Z FAIL: test (__main__.SingleHostAndService)
2022-01-19T00:56:24.2261835Z ----------------------------------------------------------------------
2022-01-19T00:56:24.2262102Z Traceback (most recent call last):
2022-01-19T00:56:24.2262474Z   File "./tests/scripts/thread-cert/border_router/test_advertising_proxy.py", line 173, in test
2022-01-19T00:56:24.2262954Z     self.check_host_and_service(server, client, '2001::2', 'my-service')
2022-01-19T00:56:24.2263403Z   File "./tests/scripts/thread-cert/border_router/test_advertising_proxy.py", line 254, in check_host_and_service
2022-01-19T00:56:24.2263793Z     self.assertEqual(client_service['state'], 'Registered')
2022-01-19T00:56:24.2264279Z AssertionError: 'ToRefresh' != 'Registered'
2022-01-19T00:56:24.2264493Z - ToRefresh
2022-01-19T00:56:24.2264655Z + Registered
```

Why the SRP server failed to propagate service TLV timely?

```
2022-01-19T00:56:24.1911889Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.308 [INFO]-PLAT----: > srp server enable
2022-01-19T00:56:24.1912304Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.308 [NOTE]-CLI-----: Input: srp server enable
2022-01-19T00:56:24.1912945Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.308 [INFO]-CORE----: [settings] Read SrpServerInfo {port:53535}
2022-01-19T00:56:24.1913387Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.308 [INFO]-SRP-----: [server] selected port 53536
2022-01-19T00:56:24.1913901Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.308 [INFO]-N-DATA--: Publisher: Publishing DNS/SRP service unicast (ml-eid, port:53536)
2022-01-19T00:56:24.1914587Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.308 [INFO]-N-DATA--: Publisher: DNS/SRP service - State: NoEntry -> ToAdd
2022-01-19T00:56:24.1915308Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.309 [INFO]-N-DATA--: Publisher: DNS/SRP service (state:ToAdd) in netdata - total:0, preferred:0, desired:2
2022-01-19T00:56:24.1916006Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.309 [INFO]-N-DATA--: Publisher: DNS/SRP service - State: ToAdd -> Adding
2022-01-19T00:56:24.1916768Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.309 [INFO]-N-DATA--: Publisher: DNS/SRP service (state:Adding) - update in 2934 msec
2022-01-19T00:56:24.1917236Z Jan 19 00:55:49 2526587c97fb otbr-agent[133]: 00:01:39.309 [NOTE]-CLI-----: Output: Done
2022-01-19T00:56:24.1917688Z Jan 19 00:55:50 2526587c97fb otbr-agent[133]: [WARN]-MDNS----: DNSServiceProcessResult failed: Bad Reference
2022-01-19T00:56:24.1918374Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: message repeated 7 times: [ [WARN]-MDNS----: DNSServiceProcessResult failed: Bad Reference]
2022-01-19T00:56:24.1919070Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [DEBG]-MAC-----: =========================[RX seqno=063 len=070]=========================
2022-01-19T00:56:24.1919902Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [DEBG]-MAC-----: | 41 D8 3F CE FA FF FF 02 | 00 00 00 00 0A 6E 16 7F | AX?Nz........n..
2022-01-19T00:56:24.1921231Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [DEBG]-MAC-----: | 3B 01 F0 4D 4C 4D 4C 17 | 89 00 15 0C 00 00 00 00 | ;.pMLML.........
2022-01-19T00:56:24.1922625Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [DEBG]-MAC-----: | 00 00 00 01 58 00 88 41 | 6D 1F 69 EF AB C2 B8 66 | ....X..Am.io+B8f
2022-01-19T00:56:24.1923256Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [DEBG]-MAC-----: | 4B A3 82 BA 75 DC 61 A6 | 5A F0 3C D5 67 FF 36 AE | K#.:u\a&Zp<Ug.6.
2022-01-19T00:56:24.1923849Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [DEBG]-MAC-----: | E7 AB 7D AA 42 7E .. .. | .. .. .. .. .. .. .. .. | g+}*B~..........
2022-01-19T00:56:24.1924495Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [DEBG]-MAC-----: ------------------------------------------------------------------------
2022-01-19T00:56:24.1925540Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [INFO]-MAC-----: Received IPv6 UDP msg, len:91, chksum:1789, from:166e0a0000000002, sec:no, prio:net, rss:-20.0
2022-01-19T00:56:24.1926259Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [INFO]-MAC-----:     src:[fe80:0:0:0:146e:a00:0:2]:19788
2022-01-19T00:56:24.1927112Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
2022-01-19T00:56:24.1927729Z Jan 19 00:55:57 2526587c97fb otbr-agent[133]: 00:01:47.311 [INFO]-MLE-----: Receive Advertisement (fe80:0:0:0:146e:a00:0:2,0x7000)
2022-01-19T00:56:24.1928602Z Jan 19 00:55:58 620bc3728f7b kernel: [  857.518899] backbone2: port 3(vethbb6abfe) entered disabled state
2022-01-19T00:56:24.1929115Z Jan 19 00:55:58 620bc3728f7b kernel: [  857.519484] veth8408a62: renamed from eth0
2022-01-19T00:56:24.1929565Z Jan 19 00:55:58 2526587c97fb otbr-agent[133]: [WARN]-MDNS----: DNSServiceProcessResult failed: Bad Reference
2022-01-19T00:56:24.1930347Z Jan 19 00:55:58 2526587c97fb otbr-agent[133]: 00:01:48.312 [INFO]-N-DATA--: Publisher: DNS/SRP service - State: Adding -> Added
[... cut ...]
2022-01-19T00:56:24.1797947Z Jan 19 00:55:32 2526587c97fb otbr-agent[133]: 00:01:22.240 [INFO]-N-DATA--: Received network data registration
2022-01-19T00:56:24.1798392Z Jan 19 00:55:32 2526587c97fb otbr-agent[133]: 00:01:22.240 [INFO]-N-DATA--: Sent network data registration acknowledgment
2022-01-19T00:56:24.1798848Z Jan 19 00:55:32 2526587c97fb otbr-agent[133]: [WARN]-MDNS----: DNSServiceProcessResult failed: Service Not Running
```

Created https://github.com/openthread/ot-br-posix/pull/1204 to collect more logs to help debug exactly which DNServiceRef is causing the `DNSServiceProcessResult` error.

jwhui, 2022-01-19:
@wgtdkp , thanks for looking into this!

jwhui, 2022-02-08:
[logs_7310.zip](https://github.com/openthread/openthread/files/8028184/logs_7310.zip)

jwhui, 2022-03-10:
https://github.com/openthread/openthread/runs/5497698717?check_suite_focus=true

jwhui, 2022-09-13:
Closing stale issue.

closed 2022-09-13 by jwhui