openthread/openthread #6665: Logging is printed across multiple lines

gabekassel, 2021-05-21:
**Describe the bug** 
Logging is unusually printed broken up across many lines.
```
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: ae
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: 45
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: c0
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: 01
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----:  s
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----:  1c00
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: ^M
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: Done
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: ^M
May 21 17:29:46.443 user.info otbr-agent[2080]: [INFO]-PLAT----: >
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id https://github.com/openthread/ot-br-posix/commit/f43ccf60be2bb65723c9216fed17874ca2cf79f9
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

jwhui, 2021-05-21:
This appears to be due to #6639.

abtink, 2021-05-21:
@bukepo looking at the code in `daemon.cpp` I noticed a pattern that I want to bring up:
- I think we can have a conifguration to have logging provided by the CLI app itself
- In this case, it will use `OutputFormatV` to implement base logging platform API  from `otCliPlatLogv()` and 
`otCliPlatLogLine()`
- But then in `OutputFormatV()` implementation in `daemon.cpp` we can use `otLog{Level}Plat()` to log.
- Can this then create a recursive call loop (for such a config)?
- Should we check and not use `otLog()` in `OutputFormatV()` fo such a config?

bukepo, 2021-05-28:
@abtink Yes, this can cause recursive call loop if we implement logging through CLI in posix app(currently, posix app doesn't have such support). On the other hand, logging through application library(CLI/Spinel) may not be a good design because they are different level components even they may share the same underlying transport and the application library itself may not use logging. I can submit a change in the example cli app directly using the underlying transport of CLI for logging. I guess it can make things clearer. Thoughts?

simonlingoogle, 2021-05-28:
> @abtink Yes, this can cause recursive call loop if we implement logging through CLI in posix app(currently, posix app doesn't have such support). On the other hand, logging through application [...]

It would still be problematic because it will change the CLI output. I think we can only log CLI if we are sure that they will be using different transports.

Irving-cl, 2021-05-30:
@gabekassel 

I'm trying to reproduce the issue. However I'm directly using `ot-daemon` not `otbr-agent` and I got similar things:
```
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: > ipaddr
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: fd9c:6aeb:6ace:45b7:0:ff:fe00:fc00
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: fd9c:6aeb:6ace:45b7:0:ff:fe00:4000
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: fd9c:6aeb:6ace:45b7:d694:ac65:ecb3:2d9c
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: fe80:0:0:0:90bf:caf:cd55:db99
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: Done
May 30 23:29:18 irvingcl.sha.corp.google.com ./build/posix/src/posix/ot-daemon[3931877]: [INFO]-PLAT----: 
```
What commands did you run and you got the output above?

jwhui, 2021-05-30:
@Irving-cl , we are seeing the same issue in GithHub Actions logs. For example: https://github.com/openthread/openthread/issues/6676#issuecomment-847981959

gabekassel, 2021-05-30:
> @gabekassel
>
>
>
> I'm trying to reproduce the issue. However I'm directly using `ot-daemon` not `otbr-agent` and I got similar things:
>
> ```
>
> May 30 23:29:18 irvingcl.sha.corp.google.com [...]

netdata show seems to reproduce this for me.

Irving-cl, 2021-05-31:
Back to this issue, the cause is that an output in cli is usually done by multiple calls to `OutputFormatV`. As an example, in `Interpreter::ProcessIpAddr`:
```
        const otNetifAddress *unicastAddrs = otIp6GetUnicastAddresses(mInstance);

        for (const otNetifAddress *addr = unicastAddrs; addr; addr = addr->mNext)
        {
            OutputIp6Address(addr->mAddress);
            OutputLine("");
        }
```
`OutputIp6Address` calls `OutputFormatV` once and `OutputLine` calls `OutputFormatV` twice. In `OutputFormatV` implementation in `daemon.cpp`, it calls `otLogWarnPlat` which then uses `syslog` and put each call to `OutputFormatV` as a line. This makes the logs I pasted above (causing two more empty lines after each address).

I think in `ot-daemon`, output is different from log. I mean the way we write output is different from the way we write log. (When we write logs, we don't add `\r\n` for it). If we want to log the output, I think we can add a stateful handling  in `OutputFormatV` implementation in `daemon.cpp`. We buffered the output content until we got a `\r\n` and then we call log function to log the output (as one line). Or we simply don't log the output. 

Thoughts? @bukepo @abtink @simonlingoogle

simonlingoogle, 2021-06-01:
@Irving-cl 
+1 for don't log the output.

bukepo, 2021-06-01:
> @Irving-cl
> +1 for don't log the output.

Outputting logging is for analyzing test failures on GitHub. How about guarding it with `OPENTHREAD_CONFIG_REFERENCE_DEVICE_ENABLE`.

simonlingoogle, 2021-06-01:
> > @Irving-cl
> > +1 for don't log the output.
> 
> Outputting logging is for analyzing test failures on GitHub. How about guarding it with `OPENTHREAD_CONFIG_REFERENCE_DEVICE_ENABLE`.

I think we should also keep CLI output clean for Reference Devices and it's quite normal for CLI output to write one or two characters each time.

Analyzing test failures should not justify adding so excess logging?
One can use other means to analyze test failures:
- Add local logging for failed TCs
- Check the actual CLI output

Thoughts?

abtink, 2021-06-01:
I agree with @Irving-cl and @simonlingoogle and +1 on not logging the output.

abtink, 2021-06-06:
I just submitted https://github.com/openthread/openthread/pull/6709 tor revert the change from https://github.com/openthread/openthread/pull/6639. 

It is clear the logging added in the above PR is far from desirable since the CLI output will be logged in pieces. 

To be honest, it's been rather annoying looking at ot-br logs like (for past weeks): 
```
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: > netdata show
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: Prefixes:
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: #015
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: fd1c:311b:6619:fa0::/64
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----:  paos
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----:  med
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----:  0400
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: #015
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: Routes:
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: #015
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: fd4c:390f:6f16:0::/64
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----:  s
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----:  med
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----:  0400
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: #015
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: Services:
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: #015
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: 44970 
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: 5d
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----:  
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: fd
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: 00
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: 0d
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: b8
June 6 29 09:11:11 ab2446a04978 otbr-agent[112]: [INFO]-PLAT----: 00
```

abtink, 2021-06-06:
I think the better way to support logging of CLI input/output would be to add this directly in the CLI module (as configurable feature) rather than adding it in the platform layer implementation (e.g., under posix). 
- It can be used under all platforms (it's indepdendept of platform layer impleemntation)
- The input/output will be logged under CLI region and not as platform log.
- Avoid issues with unintended recursive calls (avoid logging from a platform layer function which can itself be used to implement logging).

closed 2021-06-07 by jwhui