openthread/openthread #6673: Macro `OPENTHREAD_FTD` is not available in app code

superwhd, 2021-05-24:
**Describe the bug** 

I was doing the OpenThread API [codelab](https://openthread.io/codelabs/openthread-apis#0). At step [#9](https://openthread.io/codelabs/openthread-apis#8) it used the `OPENTHREAD_FTD` macro in the cli app to determine whether it's built for an FTD device. However, I found this macro seemed to always be 0/undefined. When I connected to the device using  `screen`, I found the `routerselectionjitter` to be 120 and the dataset was not the same as expected. I moved   `otDatasetSetActive(aInstance, &aDataset);` out of the `#ifdef` scope and the dataset became the same as expected. Therefore I think the cause was that `OPENTHREAD_FTD` is not available in app code.

simonlingoogle, 2021-05-24:
Should we fix this in the same way as https://github.com/openthread/openthread/pull/6474 ?

@bukepo @abtink @jwhui

abtink, 2021-05-24:
> Should we fix this in the same way as #6474 ?

Yes. Somethink like this can work

bukepo, 2021-05-25:
these two macros seems to be private. And ftd and mtd generates different link libraries, so the app build files have to specify linking to ftd or mtd, and then it can have its own macro to determine whether a feature be enabled or not.

simonlingoogle, 2021-05-26:
> these two macros seems to be private. And ftd and mtd generates different link libraries, so the app build files have to specify linking to ftd or mtd, and then it can have its own macro to determine whether a feature be enabled or not.

Seems we are defining OPENTHREAD_FTD=1 in multiple places, e.g. in openthread-ftd, openthread-cli-ftd, openthread-ncp-ftd, openthread-posix. And now seems we also need it in `ot-cli-ftd`. 

So, is it possible to make it public so that we don't need to define the same private definitions everywhere?

PR #11516 (2025-05-20): [examples] define `OPENTHREAD_FTD` and `OPENTHREAD_MTD`
Resolves #6673
closed 2025-05-20 by jwhui