openthread/openthread #6674: coap test failed

568760310, 2021-05-24:
**Describe the bug** 
I try to add some coap resource on a router, and try to get resource from BR side. 

The problem is :
1. I can get the frame sent out by BR,but can't get a response frame using wirshark.
2. BR sent out GET frame, but the source addr is always  **fdaa:bb:1::2**, even factoryreset and create a new network.

**Console/log output**  of Router 
```
> coap start
Done
> coap resource test
Done
> ipaddr
fdde:ad00:beef:0:0:ff:fe00:f400
fdde:ad00:beef:0:7414:59d9:3b85:a40a
fe80:0:0:0:90fe:7021:7fa4:986
Done
> coap request from fdaa:bb:1:0:0:0:0:2 GET
coap response sent
> version
OPENTHREAD/0.01.00; NRF52840; May 24 2021 01:49:17
```
**Console/log output**  of BR 

```
> state
leader
Done
> ipaddr
fdde:ad00:beef:0:0:ff:fe00:fc11
fdde:ad00:beef:0:0:ff:fe00:fc38
fdde:ad00:beef:0:0:ff:fe00:fc10
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:e400
fdde:ad00:beef:0:e893:bd71:8fbf:d8ab
fe80:0:0:0:689f:df:b487:184c
Done
> coap start
Done
> coap get fdde:ad00:beef:0:0:ff:fe00:f400 test
Done
> coap receive response error 28: ResponseTimeout

> version
OPENTHREAD/0.01.00; POSIX; May 21 2021 09:14:56
```

jwhui, 2021-05-24:
@568760310 , thank you for raising this issue.

Can you provide the git commit IDs you used for building both ot-cli-ftd and ot-br-posix?

I'm not sure where `fdaa:bb:1::2` comes from. On ot-br-posix, can you provide the output of `ifconfig` from the linux command line (not OpenThread CLI)?

568760310, 2021-05-25:
Yes, my ot-br-posix commit is:
```
commit f43ccf60be2bb65723c9216fed17874ca2cf79f9 (HEAD -> main, origin/main, origin/HEAD)
```

here is my output of ifconfig:
```
pi@raspberrypi:~/ot-br-posix $ ifconfig 
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.100.6  netmask 255.255.255.0  broadcast 192.168.100.255
        inet6 fe80::ba27:ebff:fe8b:dc95  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8b:dc:95  txqueuelen 1000  (Ethernet)
        RX packets 17475  bytes 1526357 (1.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 26514  bytes 6121660 (5.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 105  bytes 8360 (8.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 105  bytes 8360 (8.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

nat64: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 192.168.255.1  netmask 255.255.255.255  destination 192.168.255.1
        inet6 fdaa:bb:1::2  prefixlen 128  scopeid 0x0<global>
        inet6 fe80::f93e:aae9:c54c:9afe  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4  bytes 304 (304.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet6 fe80::689f:df:b487:184c  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:de:89:c0  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fdea:de07:60c6:2120:0:ff:fe00:fc11  prefixlen 64  scopeid 0x0<global>
        inet6 fdea:de07:60c6:2120:e893:bd71:8fbf:d8ab  prefixlen 64  scopeid 0x0<global>
        inet6 fdea:de07:60c6:2120:0:ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fdea:de07:60c6:2120:0:ff:fe00:e400  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::689f:df:b487:184c  prefixlen 64  scopeid 0x20<link>
        inet6 fdea:de07:60c6:2120:0:ff:fe00:fc38  prefixlen 64  scopeid 0x0<global>
        inet6 fdea:de07:60c6:2120:0:ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 1  bytes 56 (56.0 B)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 11  bytes 1899 (1.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

jwhui, 2021-05-25:
@568760310 , I just went through a clean setup. Unfortunately, I was unable to reproduce your issue. This is what I did.

1. Set up a Pi 4B connected to the Internet via Ethernet, using https://github.com/openthread/ot-br-posix/commit/8a2cd5b9b47176967b5899e41dd05f77ba1af9bf
```
sudo apt-get update
sudo apt-get install git
git clone https://github.com/openthread/ot-br-posix.git
cd ot-br-posix
./script/bootstrap
INFRA_IF_NAME=eth0 ./script/setup
sudo reboot

pi@raspberrypi:~ $ sudo ot-ctl
> dataset init new
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
```

2. Set up a ot-cli-ftd node using https://github.com/openthread/ot-nrf528xx/commit/d72fee5f0733461d2c23b3386434c98dc7128357
```
git clone https://github.com/openthread/ot-nrf528xx.git
cd ot-nrf528xx
./script/build nrf52840 USB_trans -DOT_COAP=ON
arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd ot-cli-ftd.hex
nrfjprog -f nrf52 --chiperase --program ot-cli-ftd.hex --reset
```

3. Add ot-cli-ftd to the network
```
> factoryreset
> dataset masterkey <masterkey from otbr>
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
```

After the devices are set up, then I did the following:

1. On ot-cli-ftd
```
> coap start
Done
> coap resource test
Done
> ipaddr
fdc4:b9d4:b56e:6083:0:ff:fe00:3800
fd54:ae61:7b68:c4fe:8523:ad39:6765:3357
fdc4:b9d4:b56e:6083:325b:f5c:20da:84a6
fe80:0:0:0:6cb6:a332:608f:f5b1
Done
> coap request from fd54:ae61:7b68:c4fe:5206:8e01:1c9d:d6f1 GET
coap response sent
```

2. On OTBR
```
pi@raspberrypi:~ $ sudo ot-ctl
> coap start
Done
> coap get fdc4:b9d4:b56e:6083:0:ff:fe00:3800 test
Done
> coap response from fdc4:b9d4:b56e:6083:0:ff:fe00:3800 with payload: 30
```

568760310, 2021-05-26:
> @568760310 , I just went through a clean setup. Unfortunately, I was unable to reproduce your issue. This is what I did.
> 
> 1. Set up a Pi 4B connected to the Internet via Ethernet, using [openthread/ot-br-posix@8a2cd5b](https://github.com/openthread/ot-br-posix/commit/8a2cd5b9b47176967b5899e41dd05f77ba1af9bf)
> 
> ```
> sudo apt-get update
> sudo apt-get install git
> git clone https://github.com/openthread/ot-br-posix.git
> cd ot-br-posix
> ./script/bootstrap
> INFRA_IF_NAME=eth0 ./script/setup
> sudo reboot
> 
> pi@raspberrypi:~ $ sudo ot-ctl
> > dataset init new
> Done
> > dataset commit active
> Done
> > ifconfig up
> Done
> > thread start
> Done
> ```
> 
> 1. Set up a ot-cli-ftd node using [openthread/ot-nrf528xx@d72fee5](https://github.com/openthread/ot-nrf528xx/commit/d72fee5f0733461d2c23b3386434c98dc7128357)
> 
> ```
> git clone https://github.com/openthread/ot-nrf528xx.git
> cd ot-nrf528xx
> ./script/build nrf52840 USB_trans -DOT_COAP=ON
> arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd ot-cli-ftd.hex
> nrfjprog -f nrf52 --chiperase --program ot-cli-ftd.hex --reset
> ```
> 
> 1. Add ot-cli-ftd to the network
> 
> ```
> > factoryreset
> > dataset masterkey <masterkey from otbr>
> Done
> > dataset commit active
> Done
> > ifconfig up
> Done
> > thread start
> Done
> ```
> 
> After the devices are set up, then I did the following:
> 
> 1. On ot-cli-ftd
> 
> ```
> > coap start
> Done
> > coap resource test
> Done
> > ipaddr
> fdc4:b9d4:b56e:6083:0:ff:fe00:3800
> fd54:ae61:7b68:c4fe:8523:ad39:6765:3357
> fdc4:b9d4:b56e:6083:325b:f5c:20da:84a6
> fe80:0:0:0:6cb6:a332:608f:f5b1
> Done
> > coap request from fd54:ae61:7b68:c4fe:5206:8e01:1c9d:d6f1 GET
> coap response sent
> ```
> 
> 1. On OTBR
> 
> ```
> pi@raspberrypi:~ $ sudo ot-ctl
> > coap start
> Done
> > coap get fdc4:b9d4:b56e:6083:0:ff:fe00:3800 test
> Done
> > coap response from fdc4:b9d4:b56e:6083:0:ff:fe00:3800 with payload: 30
> ```

@jwhui 
Thanks for your test，after I rebuild my BR， I was unable to reproduce this issue too.

closed 2021-05-26 by 568760310