openthread/openthread #6676: [github-actions] MATN_04_MulticastListenerTimeout.py fails intermittently

jwhui, 2021-05-24:
[logs_8921.zip](https://github.com/openthread/openthread/files/6535244/logs_8921.zip)

superwhd, 2021-05-25:
The error that caused the failure:
```
2021-05-24T21:40:09.9670059Z ======================================================================
2021-05-24T21:40:09.9670733Z ERROR: test (__main__.MATN_04_MulticastListenerTimeout)
2021-05-24T21:40:09.9671861Z ----------------------------------------------------------------------
2021-05-24T21:40:09.9672510Z Traceback (most recent call last):
2021-05-24T21:40:09.9673790Z   File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/thread_cert.py", line 282, in tearDown
2021-05-24T21:40:09.9674811Z     self._verify_packets(test_info_path)
2021-05-24T21:40:09.9676119Z   File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/thread_cert.py", line 310, in _verify_packets
2021-05-24T21:40:09.9677098Z     self.verify(pv)
2021-05-24T21:40:09.9678246Z   File "./tests/scripts/thread-cert/border_router/MATN_04_MulticastListenerTimeout.py", line 154, in verify
2021-05-24T21:40:09.9679435Z     pv.verify_attached('Router_1', 'BR_1')
2021-05-24T21:40:09.9680844Z   File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/pktverify/packet_verifier.py", line 192, in verify_attached
2021-05-24T21:40:09.9682094Z     src_pkts.filter_mle_cmd(MLE_CHILD_ID_REQUEST).must_next()  # Child Id Request
2021-05-24T21:40:09.9683577Z   File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/pktverify/packet_filter.py", line 233, in must_next
2021-05-24T21:40:09.9684834Z     raise errors.PacketNotFound(self.index, self._stop_index)
2021-05-24T21:40:09.9685825Z pktverify.errors.PacketNotFound: ((0, 0), (154, 154))
2021-05-24T21:40:09.9686405Z 
2021-05-24T21:40:09.9686786Z ======================================================================
2021-05-24T21:40:09.9688012Z FAIL: test (__main__.MATN_04_MulticastListenerTimeout)
2021-05-24T21:40:09.9689061Z ----------------------------------------------------------------------
2021-05-24T21:40:09.9689716Z Traceback (most recent call last):
2021-05-24T21:40:09.9690902Z   File "./tests/scripts/thread-cert/border_router/MATN_04_MulticastListenerTimeout.py", line 99, in test
2021-05-24T21:40:09.9691995Z     self.assertTrue(br1.is_primary_backbone_router)
2021-05-24T21:40:09.9695698Z   File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/node.py", line 1074, in is_primary_backbone_router
2021-05-24T21:40:09.9697279Z     return self.get_backbone_router_state() == 'Primary'
2021-05-24T21:40:09.9698971Z   File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/node.py", line 1070, in get_backbone_router_state
2021-05-24T21:40:09.9701365Z     return self._expect_result(states)
2021-05-24T21:40:09.9702884Z   File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/node.py", line 588, in _expect_result
2021-05-24T21:40:09.9703883Z     assert len(results) == 1, results
2021-05-24T21:40:09.9704400Z AssertionError: []
2021-05-24T21:40:09.9704724Z 
```

The error is the same as it in https://github.com/openthread/openthread/issues/6677. Apparently it means the OpenThread CLI returned an empty result (or timed out?) when running the command `bbr state`. I've tried running the test locally for quite some times but couldn't reproduce the failure. I'll dig into the CLI code to see if I can find the root cause.

simonlingoogle, 2021-05-25:
There seems to be some strange outputs intervened in the command output:
```
2021-05-24T21:40:09.8608950Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: > state
2021-05-24T21:40:09.8610117Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: leader
2021-05-24T21:40:09.8611072Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: #015
2021-05-24T21:40:09.8612230Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: Done
2021-05-24T21:40:09.8613109Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: #015
2021-05-24T21:40:09.8614333Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: > 
2021-05-24T21:40:09.8615497Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: > bbr state
2021-05-24T21:40:09.8617202Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: Primary
2021-05-24T21:40:09.8618154Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: #015
2021-05-24T21:40:09.8619305Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: Done
2021-05-24T21:40:09.8620609Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: #015
2021-05-24T21:40:09.8621477Z May 24 21:39:55 89b4be8f2046 otbr-agent[115]: [INFO]-PLAT----: > 
```

 Maybe we can resolve this issue when the output is cleansed.

jwhui, 2021-05-25:
The log output issue may be due to #6665.

/cc/ @bukepo

superwhd, 2021-06-15:
IIUC this issue has been fixed by https://github.com/openthread/openthread/pull/6709.

closed 2021-06-15 by superwhd