openthread/openthread #6680: How to solve this “DNS response for ipv6.google.com - Error 28: ResponseTimeout”

senthil-nag, 2021-05-25:
How to solve response timeout problem?. When I ping the Google public DNS IP address, it's not ping the address. After I give "ipaddr" in cli command it is showing this:

    ping 64:ff9b::0808:0808
    Done ipaddr
    fd11:1111:1122:0:0:ff:fe00:3800
    fd11:22:0:0:e683:e5ea:2d2d:648d
    fd11:1111:1122:0:3a09:4230:1332:7150
    fe80:0:0:0:44d7:6f79:a09d:531d
    Done dns resolve ipv6.google.com
    DNS response for ipv6.google.com - Error 28: ResponseTimeout

So I need to resolve the native IPv6 address of the given host name.

Thanks & Regards 
Senthil-nag

simonlingoogle, 2021-05-25:
Which dns server are you using. 

You can use `dns config` command to show dns configurations.

jwhui, 2021-05-25:
@senthil-nag , to ping a NAT64 address, make sure you have a default route configured. On OTBR, you can do this with the following command:
```
prefix add fd11:22::/64 paros
netdata register
```
I just tested this and was able to perform the following:
```
> ping 64:ff9b::0808:0808
16 bytes from 64:ff9b:0:0:0:0:808:808: icmp_seq=1 hlim=113 time=33ms
```

To resolve `ipv6.google.com`, the default resolver address is `2001:4860:4860:0:0:0:0:8888`, however this assume that OTBR has public IPv6 internet connectivity. If OTBR does not have global IPv6 internet connectivity, then you can leverage DNS64 on OTBR. However, you need to set the DNS resolver address to one of OTBR's IPv6 addresses. You can do this by appending the address to the command:
```
> dns resolve ipv6.google.com fdc4:b9d4:b56e:6083:0:ff:fe00:ec00
DNS response for ipv6.google.com. - 2607:f8b0:4005:807:0:0:0:200e TTL:299 
Done
```
Where `fdc4:b9d4:b56e:6083:0:ff:fe00:ec00` was the IPv6 address of my OTBR.

More documentation on the CLI commands here:
- [dns](https://github.com/openthread/openthread/tree/main/src/cli#dns-config)
- [prefix](https://github.com/openthread/openthread/tree/main/src/cli#prefix)
- [netdata](https://github.com/openthread/openthread/blob/main/src/cli/README_NETDATA.md)

senthil-nag, 2021-05-26:
Hi
Thanks for our response.
 Sorry, I am new for open thread border router, How to setup 'dns config' in cli command.
Could you please tell me that thing? 
> Which dns server are you using.
> dns resolver.

> You can use `dns config` command to show dns configurations.
It's giving command is not found "dns config".

Regards
Senthil-nag

jwhui, 2021-05-26:
@senthil-nag , I see that you are using `wpantund` and not the latest main branch of [openthread/ot-br-posix](https://github.com/openthread/ot-br-posix). Do you need to use `wpantund`?

If you haven't already, I suggest following the [Thread Border Router Codelab](https://openthread.io/codelabs/openthread-border-router#0).

senthil-nag, 2021-05-28:
Hi 
 Sorry for the late reply. I am using  nRF Nordic semiconductor software, So I have to use wpantund for NCP device.
And I get the NCP file in "nRF5 SDK for Thread and Zigbee v4.0.0"
https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.thread_zigbee.v1.0.0/thread_border_router.html?cp=8_7_5_1_0_8

jwhui, 2021-05-28:
@senthil-nag , note that nRF5 SDK for Thread and Zigbee v4.0.0 is relatively old. If possible, you should be looking at using [nRF Connect SDK](https://www.nordicsemi.com/Software-and-tools/Software/nRF-Connect-SDK).

When using the Nordic SDKs for Thread devices, you can still leverage the latest OpenThread Border Router. For setting up an OpenThread Border Router, I would recommend following one of the following:

- [OpenThread Border Router Build and Configuration
](https://openthread.io/guides/border-router/build)
- [Thread Border Router - Bidirectional IPv6 Connectivity and DNS-Based Service Discovery](https://openthread.io/codelabs/openthread-border-router#0)

jwhui, 2021-06-19:
Closing stale issue.

closed 2021-06-19 by jwhui